robotology/mesh-support #1653: Crane stop to work

lrapetti, 2023-10-12:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

I don't know if this is the right place to open this issue, but it seems that from time to time the crane stop to work. In particular, it is still able to open and close the legs, but it cannot be moved up and down.

In order to fix the issue either we unmount and mount back the upper part of the crane from the base, or we simply try to move a bit the pole without unmouting.

### Detailed context

This happened for the first time at ICRA2023, and then it happened a few times in the lab (some of the people that experienced this issue @GiulioRomualdi @paolo-viceconte @mebbaid @isorrentino @fbiggi).

lrapetti, 2023-10-12:
Consider that the crane was bought less then a year ago, so it might be still in warranty.

AntonioConsilvio, 2023-10-12:
cc @maggia80 @Fabrizio69

maggia80, 2023-10-13:
We are in contact with the supplier

GiulioRomualdi, 2023-10-13:
We had to switch to the old crane since the new one is not working

sgiraz, 2023-11-10:
Hi @GiulioRomualdi, 

After the intervention of @fbiggi, the crane of `ergoCub1.0 S/N:000` has been fixed!
Closing.

closed 2023-11-10 by sgiraz

GiulioRomualdi, 2023-11-10:
Great! Thank you 😊