robotology/mesh-support #1654: Two cables disconnected from the connector during calibration

gabrielenava, 2023-10-12:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

Two electronic cables disconnected again during the calibration. One is the same cable of yesterday, (right leg knee), while the other is the right leg ankle pitch motor.

### Detailed context

cc @davidegorbani @DanielePucci

AntonioAzocar, 2023-10-13:
I fixed the RLS3 cable of the right leg knee and the RLS4 cable of the right ankle pitch.
There are still problems with the AMO of the right knee, but it shouldn't be a wiring problem. So, today we'll continue to work on it

RLS3

[image]

RLS4

[image]

cc @gabrielenava

gabrielenava, 2023-10-13:
thank you @AntonioAzocar I have updated the title of the issue in view of the problem that emerged with the knee encoder after fixing the cable!

DanielePucci, 2023-10-17:
Also here I would add glue to the connector @AntonioConsilvio @sgiraz

AntonioConsilvio, 2023-11-10:
Hi @gabrielenava, the right knee joint continued to fail because of the EMS board to which it was connected (EB12).

In fact the channel to which the AMO was connected stopped working (Connector P7), fortunately the EMS board had other free SPI channels to which it is possible to connect the AMO.

In fact, I connected the AMO board to Connector P11.

The robot has been tested and it is working properly now!

cc @sgiraz @HosameldinMohamed

closed 2023-11-10 by sgiraz