robotology/mesh-support #1655: Robot does not turn on anymore

elandini84, 2023-10-12:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

The robot does not turn on anymore. It happened abruptly and we do not know why. 

### Detailed context

We were testing an application involving autonomous navigation and even if the battery was on 90% when we disconnected the power supply, everything turned off after less than 40 minutes. After that it did not turn back on even after plugging the power supply back in. 

### How does it affect you?

We need to test our application in the next weeks since on the 18th we of October are supposed to perform a demo.

sgiraz, 2023-10-12:
Hi @elandini84,

We are currently experiencing a high volume of work, so we cannot guarantee a resolution to your issue in time for your demo.
In order to help you more efficiently we need further details. Could you upload the log of the `yarprobotinterface` when the failure happened? Did you smell something burning? What is the status of the power LEDs?

>**Note**
>When you plan a demo I suggest you to inform all `icub-tech` through the official outlook shared calendar in order to allow us to organize better our activities in accordance with all the demo events. At the moment I cannot see any demo for October 18th.

cc @Fabrizio69 @AntonioConsilvio @Gandoo

elandini84, 2023-10-13:
Hi @sgiraz,
I do not remember smelling anything burning. The power LEDs are off. Nothing was turning on yesterday evening (around 5 pm) even when attached to the power supply.
Thanks for the note about the calendar. I'll make sure to have the demo listed there  as well. Could you send me a link for the shared calendar? I cannot see it in outlook.
Apart from the demo of the 18th (which came up just yesterday) We have the demo for "Festival della scienza" on the 2nd and the 3rd of October that we need to properly prepare and debug in the following days. That is mandatory for us.
I am sorry for the inconvenience but we really need ti have the robot up and running as soon as possible.
Today I am working from home but @fbrand-new is going to be there if you need one of us for a better explanation of the issue.

sgiraz, 2023-10-16:
Hi @elandini84,

We are planning the maintenance activity today.

> Could you send me a link for the shared calendar? I cannot see it in Outlook.

It is a shared calendar. You should see it under `Others Calendar`:

[image]

If you don't see the calendar you can simply notify @Fabrizio69 in time and he will add the event to the calendar

cc @lornat75

elandini84, 2023-10-16:
The robot works again as expected.
Thanks a lot.

@lornat75 @sgiraz

lornat75, 2023-10-18:
Thanks @sgiraz!!