robotology/mesh-support #1657: Neck tendon broken while running the robot interface on the pole

GiulioRomualdi, 2023-10-13:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

We run the robot interface on the pole and one cable of the neck got broken.

### Detailed context

This is probably due to the fact that the neck calibrates after the torso and the neck got whiplash

GiulioRomualdi, 2023-10-13:
cc @isorrentino

SimoneMic, 2023-10-13:
[image]
I've also noticed that the belt in the picture is misaligned and damaged

GiacomoVassallo13, 2023-10-17:
Hi. the damnaged cable hase been replaced wich a new one. The old cable was most damnaged bacause the tendon it slipped into the space between the pulley and the bearing. The belt it' s okay. It is misaligned and worn because the belt is cut during the assembly phase.

cc: @sgiraz @Fabrizio69 @AntonioConsilvio

sgiraz, 2023-10-18:
Well done @GiacomoVassallo13! 

Closing as solved!

closed 2023-10-18 by sgiraz