robotology/mesh-support #1660: Calibration type 10 without AMO not working

lrapetti, 2023-10-16:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

GIven the issue on the robot encoder (https://github.com/robotology/icub-tech-support/issues/1658) we tried to run the special calibration type 10 disabling the AMO sensor (as documented in https://github.com/icub-tech-iit/documentation/pull/321).

In order to do so, what we did was to replace `amo` with `none` for the first joint in https://github.com/robotology/robots-configuration/blob/0891fbec0448283c5f62fe91f6b7c4e98b4cac8e/ergoCubSN000/hardware/motorControl/right_leg-eb6-j0_3-mc_service.xml#L39. 

### Detailed context

The robot does not properly calibrate and the joint goes to idle as shown in the video:

https://github.com/robotology/icub-tech-support/assets/35487806/4d76c941-9491-4385-b754-3db19a54405a

Here is the log of the yarprobotinterface

[log_ergocub-torso_yarprobotinterface_3015_calibration.txt](https://github.com/robotology/icub-tech-support/files/12915309/log_ergocub-torso_yarprobotinterface_3015_calibration.txt)

From the yarpmotorgui, we notice that the joint measurement were not properly calibrated.

lrapetti, 2023-10-16:
cc @pattacini @ale-git @DanielePucci @CarlottaSartore @GiulioRomualdi

AntonioConsilvio, 2023-10-17:
Hi @lrapetti, talking with @ale-git, the problem is due to the EMS firmware that when using only the motor encoder to calibrate enables a timer of only one second, if the joint does not calibrate within that second, it goes into idle.

@ale-git is probably already working on this, however it is no longer necessary to disable the AMO as it has been fixed, see issue:

- https://github.com/robotology/icub-tech-support/issues/1658

GiulioRomualdi, 2023-10-18:
If the calibration type 10 is not working I would avoid to close the issue

sgiraz, 2023-10-18:
As written by @AntonioConsilvio, @ale-git is already working on it. I put the issue in review waiting for feedback from @ale-git.

sgiraz, 2023-11-10:
Hi @ale-git, do you have any update on this?

sgiraz, 2023-11-14:
> This version has been merged yesterday and works: https://github.com/robotology/icub-firmware-build/blob/devel/ETH/EMS/bin/application/ems.hex

Nice! 🚀
@lrapetti Let's flash this FW on all the EMS boards of the robot.

I'll be waiting for feedback.

cc @GiulioRomualdi @AntonioConsilvio

lrapetti, 2023-11-14:
Great!

> @lrapetti Let's flash this FW on all the EMS boards of the robot.

I have no plan to use the robot before it leaves the lab, are you going to handle the update of the board?

AntonioConsilvio, 2023-11-23:
Hi! Me and @GiulioRomualdi updated the firmware of the EMS and the robot works properly! Closing.

closed 2023-11-23 by AntonioConsilvio