robotology/mesh-support #1661: Fault not working

lrapetti, 2023-10-16:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Fault did not work this afternoon (this morning was working). Since it was also noticed in https://github.com/robotology/icub-tech-support/issues/1633, and then it started working again without intervention, it is possible that it is the same issue that appears occasionally. 

### Detailed context

Noticed during the tests of @ale-git.

sgiraz, 2023-10-18:
After a F2F with @AntonioConsilvio it seems that the problem comes and goes at random.
Actually the fault is working but We need to monitor the problem closely.

cc @AntonioAzocar @Gandoo @Fabrizio69 @AntonioConsilvio

AntonioConsilvio, 2023-11-23:
Hi @lrapetti, since the opening of this issue, the problem hasn't been observed anymore I would proceed closing this issue.

If the problem will appear again, we will reopen it.

cc @sgiraz @Gandoo @Fabrizio69

closed 2023-11-23 by AntonioConsilvio