robotology/mesh-support #1662: Left ankle roll motor moving

GiulioRomualdi, 2023-10-16:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The left ankle roll motor seems to move when pushed.

### Detailed context

This is the linked video. If you enable the audio, you can listen to the sound produced by the ankle.
 
https://github.com/robotology/icub-tech-support/assets/16744101/98e182d4-5476-494a-a0c2-fe72623a6714

### Additional context

It occurred in the past, as evidenced by: https://github.com/robotology/icub-tech-support/issues/1503.

### How does it affect you?

It would be nice to have it fixed before the next demo: https://github.com/ami-iit/lab-events-demos/issues/279

cc @DanielePucci @S-Dafarra @lrapetti @me

AntonioConsilvio, 2023-10-17:
I found these screws very loose:

https://github.com/robotology/icub-tech-support/assets/114915464/89884211-893a-436f-b1cb-e172c0d539f1

I tightened them and the problem was solved, however it might be useful to schedule an intervention to put loctite in all those screws to prevent the problem from recurring.

Anyway, the joint is now working properly.

cc @sgiraz @Fabrizio69 @fbiggi

sgiraz, 2023-10-18:
Hi @AntonioConsilvio, probably the `Loctite` alone will not be enough to fix the problem. As discussed F2F we know that `ergoCub1.1-S/N:001` had an upgrade on the material's screws. 

As of now I close the issue as solved and if necessary we'll try to put the Loctite tomorrow.

cc @fbiggi

closed 2023-10-18 by sgiraz