robotology/mesh-support #1663: Broken speaker

S-Dafarra, 2023-10-18:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The speaker works terribly. Opening the cover with @AntonioConsilvio, we noticed it is broken.was

### Detailed context

It seems that the plastic enclosure is broken, making the diaphragm move in an uncontrolled way. Hence, the bad sound quality. We temporarily fixed that with some duck tape.

sgiraz, 2023-10-18:
Hi @S-Dafarra,

This afternoon @AntonioConsilvio and @Gandoo repaired the speaker. Afaik now there is a 2 sided connector (pcb?) of the mic the was detached. Now, @AntonioConsilvio applied a piece of tape to secure the connector in place.

The problem should be solved. Let me know.

S-Dafarra, 2023-10-19:
> Hi @S-Dafarra,
> 
> This afternoon @AntonioConsilvio and @Gandoo repaired the speaker. Afaik now there is a 2 sided connector (pcb?) of the mic the was detached. Now, @AntonioConsilvio applied a piece of tape to secure the connector in place.
> 
> The problem should be solved. Let me know.

Hi @sgiraz, thanks a lot! If you want to test it on your own, if you connect in ``ssh`` to ``ergocub-head`` you can launch ``test-speaker``, that will play something in loop. Otherwise, feel free to close this :wink:

sgiraz, 2023-11-07:
Today @AntonioConsilvio tested the speaker and he confirms that it works correctly.

>**Note**
>The speaker currently mounted on ergoCub1.0 S/N:000 has been recently replaced with a new one by @Gandoo 

Closing.

closed 2023-11-07 by sgiraz