robotology/mesh-support #1664: yarprobotinterface not running after thumb proximal tendon replacement

pabloreyesrobles, 2023-10-19:
### Robot Name 🤖

iCubValparaiso01 S/N:048

### Request/Failure description

After left thumb proximal tendon replacement yarprobotinterface throw a warning and error in the communication of the BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL and CAN2.

### Detailed context

We followed the instruction guide for the replacement of the left thumb proximal tendon, where according to the documentation we had to to disconnect the lower arm. Once we replaced the tendon cable and reassembly the arm, we were not able to follow with the calibration process due to errors after running yarprobotinterface. It seems that it is related to the CAN2 bus of the EMS EB1 board according to what we see in the logs (attached below). We disassembly the arm to check if something was disconnected or damaged but we did not found any port or cable of the boards under the arm visually in that state, and by following the icub_wiring logic we checked that the CAN2 bus of the EB1 board was only connected to the LEFT SHOULDER FT SENSOR through the 1N2 cable, but it did not seemed damaged.

### Additional context

Logs after running yarprobotinterface:
[log_icub-head_yarprobotinterface_1758.log](https://github.com/robotology/icub-tech-support/files/13048219/log_icub-head_yarprobotinterface_1758.log)

Any additional information, picture or logs please let me know
cc @mjescobar @Rlanasg

sgiraz, 2023-10-26:
Hi @pabloreyesrobles,

We are processing your support request.
In the meantime, Could you provide us a screenshot of the `Firmwareupdater` where we can see what boards are discovered on both over `ETH` and over `CAN,` please?

>**Note**
>In order to discover the CAN board connected under the `EB1` (10.0.1.1), with the Firmwareupdater open, you have to select the EB1 board, click on `Force ETH Maintenance` button and wait 5 seconds. Then you have to select the `EB1` board again checking the checkbox of its row and then click on `Discover` button. If everything is ok you should see all the `CAN` boards connected under `EB1.`

mrojasanchez, 2023-10-26:
Hello @sgiraz, I'm also working with the iCubValparaiso01's group.
Here are the screenshots from Firmwareupdater:

[image]
[image]

Also, could you add me to the robotology group so I can receive messages too?

sgiraz, 2023-10-27:
Hi @mrojasanchez,

Thank you for the pictures. `FirmwareUpdater` displays all the `2FOC` boards connected under the `EB1` board, but it does not discover the `strain2` board connected to its `CAN2` interface (see below).

[image]

We can find further information within the log file posted in the comment above:

```shell
 11,599624 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3) @ 330s 79m 457u: CAN discovery after 3003 ms has detected 1 missing eobrd_strain2 boards in CAN2:
 11,599644 <ERROR> 1 of 1: missing eobrd_strain2 BOARD 10.0.1.1:CAN2:13
```
I'm pretty sure that if you try to perform the same steps described in the [comment above][1] for EB3 (1.0.1.3 right-arm) boards, you will see both the `2foc` boards and the `strain2`.

> We disassembly the arm to check if something was disconnected or damaged but we did not found any port or cable of the boards under the arm visually in that state, and by following the icub_wiring [...]

The problem may have occurred during this operation. Even though nothing appears to be damaged, I recommend checking the connections between the FT sensor (strain2 board) and the EB1 board with a digital multimeter to ensure the electrical continuity.

[1]: https://github.com/robotology/icub-tech-support/issues/1664#issuecomment-1781333755

mrojasanchez, 2023-11-05:
Hi @sgiraz, after measuring the CAN2 connections between EB1 and Strain2, we found that the connection between the J_STRN pin and the 1N2_0 red wire was broken, but due to the silicone glue, it was hard to spot. Thankfully, after repairing that connection, the Strain2 sensor was recognized by Firmware Updater. 
Thanks for your guidance!!

sgiraz, 2023-11-06:
Hi @mrojasanchez, 

I'm so glad to hear that you were able to solve the problem!
I'm confident that the issue can be closed now. 
Feel free to re-open if necessary.

closed 2023-11-06 by sgiraz