robotology/mesh-support #1665: The battery charger of the lifting device of the robot is not working anymore

gabrielenava, 2023-10-20:
### Robot Name 🤖

iCubGenova04 (iRonCub1) S/N:031

### Request/Failure description

the lifting device assigned to the robot run out of battery. The battery charger is broken, the light does not turn on and the battery does not charge.

### Detailed context

the robot will leave next week for a demo, and we need the lifting device to move it.

cc @DanielePucci

maggia80, 2023-10-20:
@gabrielenava did you search for another battery charger? I guess we should have a spare one. Are you referring to the Molyft 150?

gabrielenava, 2023-10-20:
The crane and battery charger are these one:

[image]
[image]
[image]

Specifically for the demo, I was planning to take the iRonCub3 crane. But if there is a spare battery charger I am still interested!

AntonioConsilvio, 2024-12-18:
Hi @gabrielenava, unfortunately we don't have any spare parts or possibilities to do maintenance on the crain.

Did you have any other questions or requests regarding this issue or do you think it can be closed?

gabrielenava, 2024-12-19:
>Hi @gabrielenava, unfortunately we don't have any spare parts or possibilities to do maintenance on the crain.
>
>Did you have any other questions or requests regarding this issue or do you think it can be closed?

hello @AntonioConsilvio, in the end I am trying to purchase a new battery charger and resolve the problem in that way. I'll close this issue!

cc @HosameldinMohamed @Giulero

closed 2024-12-19 by gabrielenava