robotology/mesh-support #1668: Left thumb oppose encoder readings are not valid

xEnVrE, 2023-10-30:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

After #1647, @AntonioAzocar noticed that the left thumb oppose of the robot was not calibrating and the encoder readings were strange and out of scale.

### Detailed context

The following screenshot, taken from #1647, shows the status of the thumb oppose joint after the failed calibration:

[image]

AntonioConsilvio, 2023-11-07:
Hi @xEnVrE, there was a calibration problem on the thumb proximal.

I recalibrated the joint and tested the robot.

It works properly now!

https://github.com/robotology/icub-tech-support/assets/114915464/ca54b3bc-b22c-4fa0-aa7b-25779ff5039c

Remember to update the configuration files, here the PR:

- https://github.com/robotology/robots-configuration/pull/585

cc @sgiraz

closed 2023-11-07 by AntonioConsilvio

xEnVrE, 2023-11-07:
Thanks a lot @AntonioConsilvio !