robotology/mesh-support #1669: Neck tendons broken

S-Dafarra, 2023-10-30:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During a demo, the robot fell while walking and one of the neck tendons got snapped

### Detailed context

When falling, the ropes connecting to the crane make the robot twist. This leads the ropes to entangle a little bit with the robot head, which ends up holding the robot's weight. In this configuration, the tendons snap.

When it happened, we did not have time to fix it. Moreover, we run out of tendons.

sgiraz, 2023-10-30:
Hi @S-Dafarra, thanks for you reporting!
@GiacomoVassallo13 is going to repair the neck tendons in the next few days.

GiacomoVassallo13, 2023-10-31:
Hi @sgiraz @S-Dafarra  the neck tendon has been replaced.

cc: @Fabrizio69

sgiraz, 2023-10-31:
Thank @GiacomoVassallo13!

 @S-Dafarra, I'll wait for your feedback before proceeding to close this issue.

S-Dafarra, 2023-10-31:
Ho @sgiraz, I think you can proceed. In case I will open a new issue!

sgiraz, 2023-10-31:
Ok, thanks! 
Closing.

closed 2023-10-31 by sgiraz