robotology/mesh-support #1670: Right thumb tendon broken

xEnVrE, 2023-10-30:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

During some video shooting, one of the tendons of the right thumb brake.

### Detailed context

After running the `yarprobotinterface` we noticed that the thumb was not moving anymore. Later on, thanks to @AntonioConsilvio, we realized that one of the tendons of the right thumb was broken. Not sure about the exact DoF though.

sgiraz, 2023-11-10:
Maintenance activity scheduled for next Monday (13th of November) from 9:00 to 11:00

cc @GiacomoVassallo13

GiacomoVassallo13, 2023-11-13:
Hi. Today i replaced the broken thumb oppose tendon. @xEnVrE  I attend a feedback.

cc: @Fabrizio69 @sgiraz @AntonioConsilvio

xEnVrE, 2023-11-13:
Thanks @GiacomoVassallo13 

I'll test it tomorrow.

@sgiraz @AntonioConsilvio is it safe to turn the robot as is without additional calibrations?

xEnVrE, 2023-11-14:
I have tested it, it is fine. Closing.

Thanks again

closed 2023-11-14 by xEnVrE