robotology/mesh-support #1671: Turning on motors switch causes S/D of power supply

EbruBaglan, 2023-11-02:
### Robot Name 🤖

iCubAnkara01 S/N:12

### Request/Failure description

xantrex XFR 60-46 goes into S/D when the motors' switch is turned on. 

### Detailed context

I have found an iCub in a research institution in Ankara, which is from around 20 years ago. Since I have done an internship at IIT, I thought of giving it a second life.

I have two power supplies at hand. From a previous mailing with iCub Tech, I received the following voltage and amp values to set.
- Xantrex XFR 35-35: 12/12.5V-15A
- Xantrex XFR 60-46: 40V-20A

First, I set the limits as these and then I powered the iCub. I have a switch for turning on/off the CPU and motors as seen in connection diagram. I turned on CPU and no problem. But the moment **I turned on the motors switch, xantrex XFR 60-46 goes into S/D**.

So I guess there is a problem with the motors and I am looking for a solution to spot the cause.

### Additional context

## Devices at Hand
-	iCubAnkara01 (iCub 1.x, version not exactly known)
-	Power Supply 1: xantrex XFR 35-35
-	Power Supply 2: xantrex XFR 60-46
-	iCub Laptop: DELL Latitude E4300

The model seems to be iCub 1.x as I learned from Marco Maggiali, which the documentation does not support anymore. You can find photos below to make sure. I even made a diagram of connections to spot the exact version of it, hopefully.

| [image]  | [image]  | [image]  |
|---|---|---|
| iCubAnkara01  | The single output connection  |  The connection diagram |

Note: I have not pressed any button to start icub-head, is there any button in this model? According to the documentations, there is one in the newer models, but I do not know about this model.

Thank you so much in advance and hopefully this little guy will receive its second chance in life :)

maggia80, 2023-11-17:
@EbruBaglan please check the datasheet of the power supply and please verify the OVP (overvoltage protection) [XANTREX - Power supply 2.8Kw XFR_60_46.pdf](https://github.com/robotology/icub-tech-support/files/13391873/XANTREX.-.Power.supply.2.8Kw.XFR_60_46.pdf)

Have you tried to press the S/D button? 

Could you take a picture of the back of the power supply?

EbruBaglan, 2023-11-27:
Hello @maggia80,

I checked the OVP setting and S/D functionality from the datasheet and logged my previous steps/trials and some videos of how I set the OVPs and how the problem I face is occuring on a repo.

Below are the related videos of how I set OVP and the problem in the videos from [iCubAnkara01_resurrection](https://github.com/EbruBaglan/iCubAnkara01_resurrection). 
After the second video **I pushed IN the S/D button but nada, the LED is still on even if the button is IN.**

| [image] |
|:--:|
| *Reviving the iCub - Step2.1: Setting Power Supplies* |

Here is the video of the issue:

| [image] |
|:--:|
| *Reviving the iCub - Step2.2: Powering the Robot* |

Below is the SW1 switch at the rear panel. If you need other parts shown, tell me and I will upload right away.
  | [image] |
  |:--:|
  | *SW1 at rear panel* |

According to the datasheet, S/D functionality can also be set differently from the rear panel as follows.
  | [image] |
  |:--:|
  | *S/D Function at page 63* |

And actually, in my first days of dealing with this power supply(w/o connecting to the robot), I faced an issue. S/D led was always ON, no matter what I did with the S/D button. And the way to solve it was to change the SW1-7 to the factory default (Check the [Day5 of the repo](https://github.com/EbruBaglan/iCubAnkara01_resurrection#day5) )

maggia80, 2024-01-10:
@EbruBaglan sorry for the late reply. I think you should move the SW1-7 to the previous configuration. The bigger power supply (PS)should start with 0V. Once you switch ON the 12V PS and than you turn the two selectors (pc104 and motors), the bigger PS should be enabled, powering the motor cards

EbruBaglan, 2024-01-29:
I think the order I powered the PS was the issue, and I think it is solved now, as both PS are okay (without S/D) and the leg LEDs are also on (pics below). Thank you! 

Are there other checks to make sure everything is powered without any problem? Or, if these are enough, I can ask further questions about moving the arms/legs in another issue, and we can close this issue.

  | PSs  | Leg LEDs |
  |---|---|
  | [image]  | [image] |

maggia80, 2024-01-29:
ok! great! You can go on with the tests without any problem

EbruBaglan, 2024-01-30:
Okay! Thank you and see ya!

closed 2024-01-30 by EbruBaglan