robotology/mesh-support #1672: Electrical issues on the right leg

GiulioRomualdi, 2023-11-03:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

While performing a walking experiment we noticed a tracking issue on the right hip roll. Then after few steps the joint went in over-current.  

After touching some of the cables all the hip and knee joints went in hardware fault.

### Detailed context

This is the error that we get from the interface
```
[ERROR]  from BOARD 10.0.1.6 (right_leg-eb6-j0_3), src LOCAL, adr 0, time 6036s 192m 104u: (code 0x00000013, par16 0x0006 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
```

### Additional context

This is the output of the yarprobotinterface [Uploading logger.zip…]()

S-Dafarra, 2023-11-06:
> After touching some of the cables all the hip and knee joints went in hardware fault.

For reference, this is the cable I touched 
[image]

AntonioConsilvio, 2023-11-07:
Hi @GiulioRomualdi @S-Dafarra, I tested the robot, the right leg seems to work properly. I tried touching the wires (the one signaled and others) during movement and nothing happened.

The joint works properly even at quite high ampere consumption (5A).

However, I proceeded to remove the cover and test with both tester and tweezers: the Ethernet cable connecting the EB5 board to the EB6 (E5), the AMO cable and also the LCORE cable (also tested with the jogger) of the right hip roll.

Everything seems to work properly, so we will keep the issue open for a while in case the problem recurs.

cc @sgiraz

S-Dafarra, 2023-11-08:
Possibly related https://github.com/robotology/icub-tech-support/issues/1675

AntonioConsilvio, 2023-11-16:
> Possibly related #1675

Hi @S-Dafarra, the problem was this issue was related to the one linked by you.

Since the intervention described in this [comment](https://github.com/robotology/icub-tech-support/issues/1675#issuecomment-1803352457) was done for both legs, and the problem has not recurred, I am closing the issue!

cc @sgiraz

closed 2023-11-16 by AntonioConsilvio