robotology/mesh-support #1674: iCubGenova03 – Broken Cable for the forearm skin & Problems with the motor during the index extension

MissingSignal, 2023-11-07:
### Robot Name 🤖

iCubGenova03 S/N:020

### Request/Failure description

1. The cable that connects the left forearm skin is broken. (see photo)
2. The left index motor has a problem in extension. Apparently it stops turning once a certain degree of extension is reached. (see video)

### Detailed context

The forearm skin cable should be easy to fix.
It remains to investigate the functioning of the motor of the left index finger, which stops turning once a certain degree of extension is reached.

@GiacomoVassallo13 and @AntonioConsilvio are already aware of these two problems.

### Additional context

[image]

https://github.com/robotology/icub-tech-support/assets/36140565/655cfba4-94db-450c-990c-db40808f1768

AntonioConsilvio, 2023-11-07:
cc @Matilde-Antonj

AntonioConsilvio, 2023-11-10:
Hi @MissingSignal, @GiacomoVassallo13 and I fixed the skin cable and tried to fix the index finger problem by replacing the distal tendon (and also the sheath), but unfortunately the problem persists.

During the intervention, we noticed that the distal tendon of the thumb on the left hand was broken, so we replaced that as well. 

Also during intervention, the N. 7 tendon of the left shoulder was torn at the crimp, so we replaced that as well.

Then, during the first robot start-up, the ring and pinky tendons were broken, so we replaced those as well.

Unfortunately, the problem on the index finger persists, but we may have another idea to solve this problem, you will receive instructions soon.

cc @sgiraz

sgiraz, 2023-11-14:
Just for cross reference, the same behavior is happening in:
- https://github.com/robotology/icub-tech-support/issues/1682

cc @AntonioConsilvio

Matilde-Antonj, 2024-02-06:
Hi @AntonioConsilvio,

Do you have any updates?

Thank you very much!

cc @reafrancesco @MissingSignal

sgiraz, 2024-02-29:
> Do you have any updates?

Hi @Matilde-Antonj, 

@AntonioConsilvio and I are going to plan the intervention in Erzelli next Tuesday (5th of March). We also intend to solve the problem mentioned in https://github.com/robotology/icub-tech-support/issues/1747.

cc @Fabrizio69

sgiraz, 2024-03-01:
\remind 4th March Maintenance activity scheduled for tomorrow (cc @AntonioConsilvio @Fabrizio69)

sgiraz, 2024-03-05:
Hi @MissingSignal and @Matilde-Antonj,

As per today's intervention, I would consider this issue solved.

✅ @AntonioConsilvio and I replaced the left-hand distal tendon including the proper `toflon_sheat` and We inverted the **sign** of the [PIDs for the left-index distal][3] and tuned its [calibration values][1] using the [excel file][2]. These changes were necessary because of the flipped magnets mounted on the joint.

>[!Imprtant]
> Please, note that We applied the above changes on `icub-head` live USB (icub@pc104) under `/usr/local/src/robot/robotology-superbuild/src/robots-configuration`. These changes should be merged in https://github.com/robotology/robots-configuration/tree/master through a proper PR and applied to the `build/install/` directory of the robotology-superbuild running the make command within the `build` folder.

>[!Warning]
>The general calibration order of the robot's joints is not correct. In detail, it calibrated the hands before the arms. This is a strange behavior and we have to investigate further on it. For this specific problem, We'll give you support remotely asap, tracking the activity in https://github.com/robotology/icub-tech-support/issues/1747.

If you agree, I would close this issue.
Let me know.

[1]: https://github.com/robotology/robots-configuration/blob/master/iCubGenova03/hardware/motorControl/icub_left_hand.xml#L34-L35

[2]: https://github.com/robotology/robots-configuration/blob/master/iCubGenova03/iCub_Calibration_iCubGenova03.xls

[3]: https://github.com/robotology/robots-configuration/blob/master/iCubGenova03/hardware/motorControl/icub_left_hand.xml#L62-L64

pattacini, 2024-03-06:
Relative PR:
- https://github.com/robotology/robots-configuration/pull/625

closed 2024-03-06 by pattacini