robotology/mesh-support #1675: Robot shaking

CarlottaSartore, 2023-11-08:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The robot shakes when controlled in position.
 

### Detailed context

https://github.com/robotology/icub-tech-support/assets/56030908/eabdab12-155e-451d-92c1-a33afceb6ff9

https://github.com/robotology/icub-tech-support/assets/56030908/b3e30d7e-f0c3-4c64-8969-2c75750412ab

CarlottaSartore, 2023-11-08:
C.C. @DanielePucci @lrapetti @fbiggi @S-Dafarra @GiulioRomualdi 
Maybe related to https://github.com/robotology/icub-tech-support/issues/1672

CarlottaSartore, 2023-11-08:
@carloscp3009

fbiggi, 2023-11-09:
Replacement of screws and tightening according to table
addition of LOCTITE 648 threadlocker
I await information on the correct functioning of the right leg
the same operation was performed on the left leg da @AntonioConsilvio @GiacomoVassallo13 

[image]

[image]
[image]
[image]
[image]

@maggia80 @Fabrizio69 @sgiraz

sgiraz, 2023-11-14:
Well done guys! 

I think that this issue can be closed.
Feel free to re-open if necessary.

closed 2023-11-14 by sgiraz