robotology/mesh-support #1676: Connector of the right hip pitch encoder cable broken

paolo-viceconte, 2023-11-09:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The connector of the cable of the right hip pitch encoder broke.

### Detailed context

While calibrating the robot, the cable of the right hip pitch encoder detached a few times and then its connector broke.

paolo-viceconte, 2023-11-09:
cc @DanielePucci @GiulioRomualdi @carloscp3009

AntonioAzocar, 2023-11-09:
> The connector of the cable of the right hip pitch encoder broke.

Hi @paolo-viceconte the connector mentioned in this issue is this, right?

[image]

Anyway, when can we intervernize with the AMO (cod. 6778.B) replacement?

cc @sgiraz @Fabrizio69

paolo-viceconte, 2023-11-09:
> Hi @paolo-viceconte the connector mentioned in this issue is this, right?

Yes, I meant that one (I wanted to take a picture but that piece had been detached from the robot), i.e. the connector of this cable:

[image]

> Anyway, when can we intervernize with the AMO (cod. 6778.B) replacement?

Whenever you can, from my side!

AntonioAzocar, 2023-11-09:
I replaced the AMO board with the damaged connector and now the robot is working properly again.

cc @sgiraz @paolo-viceconte

closed 2023-11-09 by AntonioAzocar