robotology/mesh-support #1677: General revision

lindalastrico, 2023-11-09:
### Robot Name 🤖

None

### Request/Failure description

As agreed with @AntonioConsilvio and @GiacomoVassallo13 we request a full revision of the robot icubHalfie01, i.e. the head-only robot of CONTACT Unit in CHT.
We need the head for an event on November 27th, so probably it's better to intervene afterward. We need the robot to participate in several projects, that's why we are asking for these fixes.

[image]

@reafrancesco @garello

### Detailed context

The known issues with such robot are:
- broken dragonfly right camera
- broken eyelids
- not functioning LEDs for facial expressions
- unavailable communication with the speaker (the device itself works, but we are unable to use it)
- broken IMU xSense
We can bring the head to CRIS for the fixing.

pattacini, 2023-11-09:
For these [warnings](https://github.com/robotology/icub-tech-support/issues/1677#issuecomment-1804113662), see https://github.com/robotology/robots-configuration/pull/587#issuecomment-1804154025.

maggia80, 2023-11-10:
@lindalastrico Unfortunately the head is obsolete and we cannot intervene. We don't have spare parts for most of the issues you are facing.  we discussed with @reafrancesco how to proceed and our options.