robotology/mesh-support #1678: Hand communication failure from board EB29

Ryo-Naohara, 2023-11-09:
### Robot Name 🤖

iCubLisboa01 S/N:004

### Request/Failure description

Hello, the issue we are having is when turning on icub relating to it's right hand.
Below is the log file for our attempt to start up icub, which shows some errors.

 @plinioMoreno
[log_icub.txt](https://github.com/robotology/icub-tech-support/files/13310387/log_icub.txt)

### Detailed context

When starting icub's routine, communication with board EB29, the board on the right side responsible for the hand, fails to connect and we can't seem to use the motors on the hand, they can't seem to be put to idle either, 8 motors in total from the hand all appear has red in yarpmotorgui. However the encoders still work since they still output values that we can see in yarpmotor gui.
We've tried checking the cables and connecters but everything seems to be seated correctly and no signs of a cut or snag wire.

### Additional context

We were trying to fix an issue with the right hands index finger proximal motor not moving with yarpmotorgui, after doing some testing to see what was wrong, the hand stopped working.

AntonioConsilvio, 2023-11-16:
Hi @Ryo-Naohara, sorry for the late reply.

Judging from the described behavior and the type of errors in the logger, the cause could be the missing of CAN connection between the EB29 board and MAIS.

I recommend that you perform the following tests:

- Start the `FirmwareUpdater` command on the iCub head (with -X because it is a graphical application), then select the first option and click Discover. 
Then select the mc4plus board with address 10.0.1.29.
Next, click on Force ETH maintenance. 
Then (again with board 10.0.1.29) click Discover. Here you can see whether the CAN board is well connected or not. (Please send a screenshot).

- Perform the same procedure for the EB28 board.

- If you cannot see any CAN board under either board, you can (even without opening the palmetto) see if the MAIS board is on (red circles) and you can also see the connector (blue circle):

[image]

- If you want you can also open the back of the hand and check the inside:

[image]

> [!Note]
> In case you open the back of the hand, I also recommend to check the continuity of the cable from MAIS to EB29 (cable name: 4N1).
> The board EB29 is on the upperarm.

In case you have any questions, please feel free to ask them!

cc @sgiraz

Ryo-Naohara, 2023-11-17:
Hello @AntonioConsilvio , thank you for the reply.
Checking the cable it seems 2 of the wires in the cable you mentioned (4N1).

Here is a photo of the connector.
[image]
And a paint diagram to show how it is disconnected.
[image]

Ryo-Naohara, 2023-11-20:
Hello, correction of pin connection, the cable 4N1 which is the one with broken cables was connected to Board 29 via a 6pin connector attached to p8 and, a 7pin connector attached to p10.

AntonioConsilvio, 2023-11-22:
Hi @Ryo-Naohara, we found the correct diagram to fix the damaged wiring.

[image]

Basically, the MAIS board takes power (red/yellow and shield+black wires) from connector P10 (7pin connector), while CAN communication is done on connector p8 (6pin connector) with Green, black and white wires.

However, the logic diagram I refer to can be found here:

https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/iCub2.5-E3.3.2_5672_Logic.pdf

In case you have any questions, please feel free to ask them!

cc @sgiraz @AntonioAzocar

AntonioConsilvio, 2023-12-07:
Hi, @plinioMoreno, try to check all the connections (5 in total) as the one indicated (see red circle 🔴) in the picture below:

[image].

Ryo-Naohara, 2024-01-23:
@plinioMoreno 

Hello @AntonioConsilvio It has been a while since my last contact, i tried to check the wires and 2 of them were broken, testing it again with the multimeter it seems they all are connecting correctly but the hand is still at fault and when the wires are moved they seem to jolt a little bit, with the log below, could you confirm that its probably still a bad connection with the wire?

[log_icub1.txt](https://github.com/robotology/icub-tech-support/files/14025345/log_icub1.txt)

AntonioConsilvio, 2024-02-05:
Hi @Ryo-Naohara, sorry for the late reply! 
The problem seem similar to the previous ones, but I need some information to determine the cause.

You could provide us with this information:

- If you can perform the first two steps of this [comment](https://github.com/robotology/icub-tech-support/issues/1678#issuecomment-1814097202) (`FirmwareUpdater`)

- Video of the fingers during robot start-up.

- Screenshot of the `yarpmotorgui` of the finger joints once the calibration is complete.

- Check from the `yarpmotorgui` that the encoders of all fingers read, by trying to move all finger joints slightly by hand. (adduction, oppose, proximals and distals joints).

cc @sgiraz

AntonioConsilvio, 2024-02-12:
Hi @Ryo-Naohara, thank you for all information!

Analysing the situation, the problem on CAN communication seems to persist.

In fact, from the `FirmwareUpdater` you should be able to view CAN boards (in this case the MAIS board) under the ETHboards 10.0.1.28-29.

The fact that the CAN communication is broken means that the finger sensors are not able to send values to the ETH boards, and so it makes sense that from the `yarpmotorgui` the joint encoders remain static.

The CAN wiring is as follows in the diagram (cable name: 4N1):

[image]

If you want to see the wiring diagram more specifically:
- https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/iCub2.5-E3.3.2_5672_Logic.pdf (page 10).

I recommend checking the wiring and testing the continuity with the tester. You can also test that there are no short circuits between the cables.

I also recommend to see if the MAIS board in the hand has any LEDs on (so as to exclude the red 5V and black GND cable among the possible faulty cables).

I also recommend making sure that the CAN cable has been connected to the EB28-EB29 boards and not to the EB27 by mistake.

Please, send us feedback and if you have any questions feel free to ask!

cc @sgiraz

Ryo-Naohara, 2024-05-15:
Hello, it seems the issue was that the there was a faulty cable connection from the hand to the EB29, even though through the multi meter seemed to be fine, there was still an issue with it, remaking the connectors seems to have fixed it, there was also another issue where the power connection from the hand to the board seems to have been faulty, as a temporary measure we connected it from another board which seems to have worked since the mais board now appears on the firmware updater.

closed 2024-05-15 by Ryo-Naohara