robotology/mesh-support #1679: Right hip roll motor goes into overcurrent fault when moving (i2t fault)

HosameldinMohamed, 2023-11-10:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

When trying to move the right hip roll joint of the robot, it goes into a hardware fault and it says it is passing the i2t limit.

### Detailed context

We checked the motor current while moving the joint and it grows to a very high value (around 20A if not wrong @davidegorbani). At this point we are not sure if the current measurement is real and the current is really high, or the maybe the measurement is wrong.

We made a short video to describe the movement, as shown below.

The left hip roll joint does the same movement without an issue.

### Additional context

The video shows another problem happening with the torso roll joint, but I will open another issue for that.

https://github.com/robotology/icub-tech-support/assets/45564317/7dc326d3-0673-4d6e-96bc-02ae206c6256

### How does it affect you?

It could block iRonCub experimental activities.

CC @gabrielenava @DanielePucci

sgiraz, 2023-11-10:
Closing as duplicated #1652 

cc @AntonioConsilvio

DanielePucci, 2023-11-10:
@sgiraz This issue **is not** a duplicate of https://github.com/robotology/icub-tech-support/issues/1652 although it has the same video. 

In fact, the title of this issue is *Right hip roll motor goes into overcurrent fault when moving (i2t fault)*, namely, we read 20 A with very small variations of the joint, and this is not normal.

CC
@maggia80

DanielePucci, 2023-11-10:
Apologies @sgiraz, I got the wrong linking in the browser due to cache memory: you are right, it is a duplicate . Closing.

closed 2023-11-10 by DanielePucci