robotology/mesh-support #1680: Torso roll joint jumps at random times to random positions

HosameldinMohamed, 2023-11-10:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The torso roll joint jumps to a random position during normal operation. The trigger of this is still unknown, it happened during the moving of the legs joints.

### Detailed context

The video below shows one time when the problem occurred.

### Additional context

The same video shows also #1679.

https://github.com/robotology/icub-tech-support/assets/45564317/2461fd7f-0785-4f2c-9a6c-1ba5cc121b7c

### How does it affect you?

It blocks iRonCub experiments.

CC @gabrielenava @DanielePucci

GiacomoVassallo13, 2023-11-14:
Hi. The faulted motor was the hip roll of right leg not the waist. 
The faulted motor has been replace wich a new one. 
I attach the photo of the replaced stator:

[image]

The optical disc was also replaced as it was worn ,and the correct reading was verified with the bench test.

Unfortunately, I would like to point out that the assembly is missing some components: in particular the front edges of the sprockets and a  piece of the belt tensioner.

[image]

This components will be added the as they become available.

The motor group was reassembled on the robot. @AntonioAzocar will take care of the electronic wiring of the new engine.

cc: @Fabrizio69 @sgiraz @AntonioConsilvio

gabrielenava, 2023-11-14:
to clarify: the issue that is tracking the hip roll problems is this one: https://github.com/robotology/icub-tech-support/issues/1652

while we opened this issue to understand why the torso roll sometimes jumps from one position to another, which is not correlated with the hip roll problem.

cc @sgiraz @Fabrizio69

GiacomoVassallo13, 2023-11-14:
Thanks @gabrielenava

HosameldinMohamed, 2023-11-15:
We started the robot and torso roll joint behaving as expected. The issue did not appear again for the whole period of the test!

AntonioConsilvio, 2023-11-15:
Just for clarity, the cable found damaged and fixed by @AntonioAzocar was the cable that connects the optical encoder (LCORE5) with the 2FOC.

Since the robot has been tested, I will proceed with closing the issue, if the problem returns feel free to reopen this issue!

cc @sgiraz @HosameldinMohamed @gabrielenava

closed 2023-11-15 by AntonioConsilvio