robotology/mesh-support #1681: Wrong right leg home position after calibration

SimoneMic, 2023-11-10:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

When homing the right leg from the yarp motor GUI the leg its held upfront, instead of its original position (see photo below)
[image]
No errors shown during the calibration, nor afterwards.

### Detailed context

I have tried launching different yarp robot interfaces, both `ergocub_wbd.xml` and `ergocub_wbd_ros2.xml` from the robot torso. The result is the same.
I am using the `robots_configuration` branch `ami_20iit` and I haven't modified any files, though there are many uncommitted and changed files.
I suspect that the right hip pitch is not homed properly, or the home configuration is not correct. Probably is a configuration file issue.
Also, while using the `WalkingModule` and preparing the robot, it gets in a wrong configuration:
[image]

### How does it affect you?

I can't walk and navigate with the robot in this state

S-Dafarra, 2023-11-10:
Hi @SimoneMic what does the hip pitch encoders measure in this position?

SimoneMic, 2023-11-10:
Hi @S-Dafarra  the `r_hip_pitch` reads 11.518
[image]

GiulioRomualdi, 2023-11-10:
As far as I know, we didn't change the positions. However, yesterday I noticed that the calibrators on the legs changed but since I didn't know who changed it I left it there. So probably this is the issue. Notice that the robot went under maintenance for https://github.com/robotology/icub-tech-support/issues/1675 so I don't know if this is related

AntonioConsilvio, 2023-11-10:
Hi @SimoneMic, the AMO has been replaced in this [issue](https://github.com/robotology/icub-tech-support/issues/1676), but it was not calibrated.

Me and @AntonioAzocar calibrated the AMO and tested the robot. 
It works properly now!

cc @sgiraz

closed 2023-11-10 by AntonioConsilvio