robotology/mesh-support #1682: [Right hand calibration issue]

ddetommaso, 2023-11-13:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During calibration the right hand closes and remains closed without opening.

### Detailed context

[log_icub-head_yarprobotinterface_1746.txt](https://github.com/robotology/icub-tech-support/files/13335215/log_icub-head_yarprobotinterface_1746.txt)

sgiraz, 2023-11-13:
> During calibration the right hand closes and remains closed without opening.

Possibly related to:
- https://github.com/robotology/icub-tech-support/issues/1674

sgiraz, 2023-11-13:
Hi @ddetommaso,

I would ask you to add additional context, if possible.
For example, did you re-calibrate some joints of the hand? 
In which situation was the problem raised?
Do you have any videos or photos that can show the problem?

ddetommaso, 2023-11-14:
Hi @sgiraz,

the issue is occurring non systematically during data collection with participants, so it's not easy to debug and report more details at the moment. Anyway, we did not re-calibrate the joints of the hand and the issue occurs usually when we turn on the robot in the morning.

Please let's schedule an intervention ASAP.

Thanks

AntonioConsilvio, 2023-11-28:
Hi @ddetommaso! I tried to test the robot multiple times, but the problem did not occur to me.

Also, I tried to check the cable that could cause that kind of error on the EB29 board (cable name: 4N1), and it was in very good condition.

I also tried touching and moving the cable during robot startup trying to prompt the event, but it didn't happen.

Another reason that could in some cases cause similar problems is the startup of the `yarprobotinterface` too fast compared to the startup of the motors.

Anyway, since it hasn't reoccurred, I would close the issue, however feel free to reopen it if necessary!

cc @sgiraz

closed 2023-11-28 by AntonioConsilvio