robotology/mesh-support #1683: Right wrist reaches the limit while calibrating and it remains there

GiulioRomualdi, 2023-11-13:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

While walking, the right wrist reached its limits unexpectedly, without being controlled by the walking controller. We attempted to reboot the robot and the motors several times, but when the YARP robot interface started, the motor exceeded the limits 

### Detailed context

To prevent damaging the motors, we immediately switched off the robot.

### Additional context

Here are images of the robot after executing the YARP robot interface, revealing the asymmetry.

[image]
[image]

### How does it affect you?

There's a demo scheduled in the upcoming days (https://github.com/ami-iit/lab-events-demos/issues/281). Additionally, it's worth noting that the robot will be departing from IIT permanently (https://github.com/ami-iit/lab-events-demos/issues/284).

cc @S-Dafarra @DanielePucci @carloscp3009 @paolo-viceconte

S-Dafarra, 2023-11-14:
I have checked the ``yarpmotorgui`` and all the wrist joints remain constantly equal to zero even when moving the joint by hand. I also started the robot with [these two parameters](https://github.com/robotology/robots-configuration/blob/003ffaa1de00901e0afee9767e4d110c633f0a57/ergoCubSN000/general.xml#L7-L8) to true and this is the output

[image]

When moving the joints by hand, the motor encoder change, but not the joint ones.

cc @Fabrizio69 @Gandoo

sgiraz, 2023-11-14:
>I also started the robot with [these two parameters](https://github.com/robotology/robots-configuration/blob/003ffaa1de00901e0afee9767e4d110c633f0a57/ergoCubSN000/general.xml#L7-L8) to true and this is the output

Hi @S-Dafarra which version of FW of AMC is currently flashed on the robot?

>I have checked the yarpmotorgui and all the wrist joints remain constantly equal to zero even when moving the joint by hand.

Probably we have to check the wires/connector of the AEA2 encoder of the wrist.

S-Dafarra, 2023-11-14:
> > I also started the robot with [these two parameters](https://github.com/robotology/robots-configuration/blob/003ffaa1de00901e0afee9767e4d110c633f0a57/ergoCubSN000/general.xml#L7-L8) to true and this is the output
> 
> Hi @S-Dafarra which version of FW of AMC is currently flashed on the robot?

101.8

Indeed, there is a mismatch between the two wrists

[image]

sgiraz, 2023-11-14:
> Indeed, there is a mismatch between the two wrists

In this case, it is normal, since it is an old FW version compared to the latest one.

### 2 possible solutions:

1. #### A "quick" swap test:
It consists of swapping the 2 AMC boards with the aim of proofing whether there is a faulty board or not. To do this test you need the same binaries currently flashed on the robot (usually within the `icub-firmware-build` folder on the robot). Swap physically the 2 boards and flash the `amc.wrist.left.hex` binary to the board mounted on the left arm and the amc.wrist.right.hex mounted to the right one.
If the problem is swapped from the right wrist to the left, it means that the board has to be replaced.

2. #### Flash the [latest FW version of the `amc`][1]
In this case, there exists only one binary called `amc.hex` to flash on both amc boards. In order to use this FW you have to perform some minor changes to the configuration files of both left and right arms:

in `ergoCubSN000\hardware\mechanicals\right_arm-eb30-j4_6-mec.xml` replace the `JOINTSET_CFG` group with the following:
```xml
    <group name="JOINTSET_CFG">
        <param name= "numberofsets"> 1 </param>
        <group name="JOINTSET_0">
            <param name="listofjoints">  0 1 2         </param>
            <param name="constraint">    ergocubwrist  </param> <!-- none for calibration -->
            <param name="param1">        20            </param>
            <param name="param2">        1             </param>
        </group>
    </group>
```

in `ergoCubSN000\hardware\mechanicals\left_arm-eb31-j4_6-mec.xml` replace the `JOINTSET_CFG` group with the following:
```xml
    <group name="JOINTSET_CFG">
        <param name= "numberofsets"> 1 </param>
        <group name="JOINTSET_0">
            <param name="listofjoints">  0 1 2         </param>
            <param name="constraint">    ergocubwrist  </param> <!-- none for calibration -->
            <param name="param1">        20            </param>
            <param name="param2">        0             </param>
        </group>
    </group>
```

>**Warning**
> The solution nr.2 hasn't been tested yet on ergoCub1.0 S/N:000, but only on the wrists of ergoCub1.1 S/N:001. It should be fine with the configuration posted above, but please pay attention.

cc @valegagge @marcoaccame @Nicogene 

  

[1]: https://github.com/robotology/icub-firmware-build/tree/devel/ETH/AMC

sgiraz, 2023-11-14:
During an F2F call with @S-Dafarra I got that the wrist was working correctly while it was walking. Then at a certain point, the wrist went out of control. In this case, it shouldn't be a FW's version problem. It is more probable that it is a HW problem since the jogger reads the AEA2 values correctly.

sgiraz, 2023-11-14:
After detaching and re-attaching the `amc` of the right arm (without changing any FW version), something seems improved. Another indicator that it may be a mechanical or electrical issue.

sgiraz, 2023-11-14:
The wrist is now under repair since one of the belts is broken:

[image]

cc @S-Dafarra @GiulioRomualdi @fbiggi @fiorisi @GiacomoVassallo13

S-Dafarra, 2023-11-16:
@maggia80 also spotted a damaged wrist encoder connector. This might have been the source cause of this issue, although the breakage of the belt might have been caused by continuous wear.

closed 2023-11-20 by sgiraz