robotology/mesh-support #1684: Right shoulder pitch cable has one broken wire

HosameldinMohamed, 2023-11-15:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We just saw that the cable that connects the optical encoder to the right shoulder motor `E1` that has one wire broken.

### Detailed context

The broken wire is the black one. We do not recall any activity that caused this issue.

The glue seems old and hardened.

### How does it affect you?

It may block iRonCub experiments.

HosameldinMohamed, 2023-11-15:
We will attempt a quick fix if we borrow the tools!

HosameldinMohamed, 2023-11-15:
From https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub3/pdf/iCub%203_10360_Upperbody%20Harness.pdf

Here's the type of connector and size (AWG):

[image]

HosameldinMohamed, 2023-11-15:
With @pillai-s we crimped the broken cable! We started the robot and joint works as expected!

The fix is temporary because the repaired wire is shorter than the others. Also no glue is applied on connector.

In general the whole cable seems short so probably if a new crimping is necessary we might need to replace it!

gabrielenava, 2023-11-30:
cc @Gandoo

Gandoo, 2023-11-30:
ciao @gabrielenava  ho ripristinato il cavo e preparato i cavi spare

[image]

[image]

[image]

cc @AntonioConsilvio @sgiraz

closed 2023-12-07 by AntonioConsilvio