robotology/mesh-support #1686: Right hip pitch AMO is not working

gabrielenava, 2023-11-15:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

During calibration, the right hip pitch knee joint faulted without being able to complete its routine. 

### Detailed context

After a quick check on the `yarpmotorgui`, we saw that the AMO encoder was frozen. We were not able to locate any visible broken cable, so we have no further details for understanding the reason of this fault at the moment.

cc @DanielePucci @HosameldinMohamed 

### Additional context

We were able to overcome the problem by disabling the AMO encoder thanks to the new firmware: https://github.com/robotology/icub-tech-support/issues/1628#issuecomment-1809838776

after disabling it, the joint calibrated with  no issues, thus further confirming that the problem was the AMO.

AntonioConsilvio, 2023-11-16:
Hi @gabrielenava, since you have decided to use only the optical sensor on all joints (see [comment](https://github.com/robotology/icub-tech-support/issues/1628#issuecomment-1812732359)), do you think interventions to fix the unused sensor would be useful or do we close the issue?

cc @sgiraz @Fabrizio69

gabrielenava, 2023-11-21:
>do you think interventions to fix the unused sensor would be useful or do we close the issue?

We still did not test much the new setting for the encoders, for now I would leave the issue opened. Maybe we can give to this intervention a low priority, then fix it in combination with the next issue we will open when something else on the leg breaks.

AntonioConsilvio, 2023-11-23:
Hi @gabrielenava, unfortunately We cannot keep the issue opened waiting that others issues happen on the legs.

If you prefer fix this problem we can schedule the intervention, otherwise We would proceed closing this issue.

cc @sgiraz

gabrielenava, 2023-11-23:
Hello @AntonioConsilvio 

at this point, I would then prefer to schedule the intervention. I think that it is better to have the robot fully working even if the hip AMO is not used in this moment.

gabrielenava, 2023-12-06:
While opening the covers for the wind tunnel activity we understood why the AMO is not working: two cables on the EMS side disconnected from the crimp. Another cable on the AMO side also disconnected from the crimp while operating for the wind tunnel. For now we don't fix the cable because we are not using the AMO. Can you replace the cable at this point when we come back from the wind tunnel (14th december)?

AntonioConsilvio, 2024-04-24:
Hi @gabrielenava! Even if not used, as decided I fixed the damaged AMO cable!

The connector was damaged on the AMO side.

closed 2024-04-24 by AntonioConsilvio