robotology/mesh-support #1688: The robot sometimes completely blacks out when in battery mode

S-Dafarra, 2023-11-16:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

It happened that during two demos, the robot suddenly shuts down completely when in battery power.

### Detailed context

This happened twice in similar conditions. The robot starts in the following state:
- it is powered on
- the battery is on and charged (>75%)
- the power supply is connected
- the battery is being charged

Then, we remove the battery charger and the power supply. After 30s or so, the robot falls down and the robot is completely off, also the display.

It is then necessary to power cycle the battery to have the robot functional again, and the battery is still charged.

### Additional context

An educated guess is that the battery BMS goes into protection. We suspect that there is an overheating issue. Indeed, the second time the robot was quite hot, but the first time we were with the robot "outside" and the whether was quite fresh.

maggia80, 2023-11-17:
yes, the BMS of the battery goes in protection. In order to recover the BMS from the fault status, a switch on and off of the battery.  
There are 4 possible faults:

- overcurrent, the max current absorption is 40A for just 50ms
[image]

 - overvoltage (>42V).  
 - undervoltage 
 - overtemperature

The problem is that we cannot read these status bits...
We can keep monitoring this issue and wait to get more insight

SimoneMic, 2024-05-20:
Hello!

Something really similar happened to `ergoCubSN001` during the tests for the demo in Florence.
After an extended time of operation on battery mode, the robot completely shuts down and restarts, like a suddend void of power.
At the moment of the power outage, the robot was (luckily) suspended on the crane and standing still, with the motors always on and controlled.

The differences that were present, albeit really similiar, with the original post were:
- The battery level was around 35%
- The robot was operating normally for an extended period of time.
- The robot display on the back of the robot was on, but both the motors and PCs were off (red leds)

@AntonioAzocar  @AntonioConsilvio were present, maybe thay can add some details that I've missed
cc @S-Dafarra

S-Dafarra, 2024-05-24:
> The differences that were present, albeit really similiar, with the original post were:
> 
> * The battery level was around 35%
> * The robot was operating normally for an extended period of time.
> * The robot display on the back of the robot was on, but both the motors and PCs were off (red leds)

It happened again today. It seems like the BAT rebooted for some reason. May it be a FW issue on the BAT side? @maggia80 @valegagge @MSECode

MSECode, 2024-05-24:
oks,
@S-Dafarra the robot was powered by battery as for previous occurrence, by power supply or by both of them?
Moreover, do we have a log of the YRI?
Another tings, bms and/or bat devices were enabled and so do we have some data collected by the TelemetryDeviceDumper?
It would be good to check those since we can maybe have issues on the bms side or even to the battery pack.
Since this happens already couple of times lately and the behavior is similar, I'm thinking that there might be issues on the battery pack.
Being the robot powered by the battery pack, I'm not sure there's something wrong with the BAT fw. If that was the case you should have similar problem even when using the power supply and more frequently.
Thus, please add here whatever data you have, so I'll look for any meaningful information.

S-Dafarra, 2024-05-24:
> @S-Dafarra the robot was powered by battery as for previous occurrence, by power supply or by both of them?

Only battery.

> Moreover, do we have a log of the YRI?

I am afraid not 😢 

> Another tings, bms and/or bat devices were enabled and so do we have some data collected by the TelemetryDeviceDumper?

For this, I am tagging @carloscp3009 and @GiulioRomualdi, but I am not sure we were logging the data. I think the battery device was active though since we the YRI throws errors about it (something like that it detected that some boards are not streaming).

> Being the robot powered by the battery pack, I'm not sure there's something wrong with the BAT fw. If that was the case you should have similar problem even when using the power supply and more frequently.

The reason why I am thinking about this is because it is happening on three different robots in very similar conditions, while it never happened on iCub3 (at least before it got 🚀  mounted on it). The probability that three battery packs are faulted in the same way seems low, and iCub3 had the same BMS. The only difference is the BAT, which is common between ergoCubs.

MSECode, 2024-05-24:
> For this, I am tagging @carloscp3009 and @GiulioRomualdi, but I am not sure we were logging the data. I think the battery device was active though since we the YRI throws errors about it (something like that it detected that some boards are not streaming).

Oks, the errors related to the fact that the boards are not streaming happens when a device is enabled in the xml configuration files but it's not streaming anything on CAN. It's generated by  the 'can monitor', which is pinging the CAN boards every 1s. 

> The reason why I am thinking about this is because it is happening on three different robots in very similar conditions, while it never happened on iCub3 (at least before it got 🚀  mounted on it). [...]

Yeh, that's reasonable. I'll dig into that to see if there's something wrong in the fw.
Anyway, since it seems that this issue is happening quite often lately and it is difficult to simulate it on a test setup could you please have the DeviceDumper collecting data during all the experiments so it's easier for us to reach the source of the problem.

S-Dafarra, 2024-05-24:
> Anyway, since it seems that this issue is happening quite often lately and it is difficult to simulate it on a test setup could you please have the DeviceDumper collecting data during all the experiments so it's easier for us to reach the source of the problem.

Makes sense! Just to be sure, which data are you interested in?

S-Dafarra, 2024-05-24:
> For this, I am tagging @carloscp3009 and @GiulioRomualdi, but I am not sure we were logging the data. I think the battery device was active though since we the YRI throws errors about it (something [...]

I checked with them and we inadvertently deleted all the logs because we ran out of HD. We did not think of keeping that one.

valegagge, 2024-05-24:
Remember to dump the data on a laptop! 😉

valegagge, 2024-05-24:
> Makes sense! Just to be sure, which data are you interested in?

We need the output of the battery devices (both for BAT and BMS), but @MSECode can add more details and ideas.

MSECode, 2024-05-24:
Exactly,
since those device ports, if enabled, are already opened when the YRI starts we can add them to your TelemetryDeviceDumper and get out the data

S-Dafarra, 2024-05-24:
> Remember to dump the data on a laptop! 😉

Unfortunately, it was the laptop to be full. We were focused on an experiment and we were logging a lot of data, and to continue with the experiment we decided to delete the past logs.

> Exactly, since those device ports, if enabled, are already opened when the YRI starts we can add them to your TelemetryDeviceDumper and get out the data

Yeah, I think that at the moment that data is not logged

S-Dafarra, 2024-06-10:
Hi all, soon we are going to have a demo on a stage without any cable. Do you think there is something that is possible to do in order to reduce the likelihood of this happening? 

Can this be related to https://github.com/robotology/icub-firmware/pull/421 since it started happening afterward?

valegagge, 2024-06-11:
Hi @S-Dafarra, today we gonna check if those changes caused the issue.

In the meantime, did you dump some data?

cc @MSECode

S-Dafarra, 2024-06-11:
> In the meantime, did you dump some data?

I could not since I was busy all week in Washington for a demo. Moreover, to the best of my knowledge, it did not appear again lately.

MSECode, 2024-06-11:
Hi @S-Dafarra, I checked out the code but it seems all fine to me. The changes done in that PR should be fine. I do not notice anything strange. Anyways, is the latest version of the BAT, i.e. 1.3.4 flashed on all the robot you are using? And I'll keep this issue monitored eventually.

S-Dafarra, 2024-10-31:
Today it happened again (on the SN000). We operated the robot in battery for a while, then we plugged back the power supply and the battery charger. After about 20 minutes, we switched off the battery charger, we switched off the power supply, and about 10 seconds later, the robot completely blacked out. The display was off and I had to power cycle also the BMS. I was reading the output of the BMS, here the last lines (FYI the period is 1s, see https://github.com/icub-tech-iit/robots-configuration/blob/0f5e570b4aaeb283d588fdb0cbb5c5b231714ccf/ergoCubSN000/wrappers/battery/battery_bms.xml#L4)
```
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
36.5 0.0 45.0 34.0 0
[... 1169 lines cut ...]
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0
35.700000762939453125 -5.30000019073486328125 45.0 34.0 0

[INFO] |yarp.os.impl.PortCoreInputUnit|/tmp/port/1| Removing input from /ergocub/battery/bms/data:o to /tmp/port/1
```

S-Dafarra, 2024-10-31:
It does not seem to be an overheating issue, since the temperature was 34 deg.

S-Dafarra, 2024-12-10:
During Humanoids, it seemed to happen also with the 001, when the battery was at 30%. The ambient temperature was also very cold.

AntonioConsilvio, 2025-02-20:
Hi! Probably the cause of the problem is that after many uses, the batteries start to wear out.

Regarding ergoCub SN: 001, we recently replaced the battery with the one removed from ergoCub SN: 002 before the upgrade.

So now ideally the situation should improve as that battery should work properly.