robotology/mesh-support #1689: Chest IMU cable pulled out from sleeve and probably disconnected

gabrielenava, 2023-11-23:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The cable of the chest IMU pulled out of the sleeve during experiments with iRonCub, close to the connection with the chest IMU. The cables inside do not look disconnected or broken, nevertheless the IMU does not start anymore. We created a workaround with and external cable and the IMU is working fine, so we suspect that the cable not only exited the sleeve but also got damaged somehow.

### Detailed context

I will add  a picture of the cable

Gandoo, 2024-01-25:
fili strappati 

[image]

Gandoo, 2024-01-25:
ciao @gabrielenava  ho ripristinato il filo nero strappato e la schermatura del cavo che era saldata sul manicotto del connettore. Vi chiedo solamente di fare attenzione nello sconnettere il connettore dalla IMU facenso forza dove indicato dalla freccia e NON sul cavo per evitare di danneggiare il cavo. 
Grazie, Buona giornata!!

[image]

[image]

cc @Fabrizio69 @DanielePucci @AntonioConsilvio @sgiraz

sgiraz, 2024-01-29:
Ciao @gabrielenava,

Have you tested the IMU? I guess that is working now, isn't it? :) 

cc @Gandoo @AntonioConsilvio @Fabrizio69

gabrielenava, 2024-01-29:
>Have you tested the IMU? I guess that is working now, isn't it? :)

ciao @sgiraz not yet, but there are some tests scheduled for tomorrow and I can verify the IMU

gabrielenava, 2024-02-06:
I have tested the robot in these days and the IMU is working fine! thanks again! we can close the issue

closed 2024-02-14 by AntonioConsilvio