robotology/mesh-support #1690: Right thumb proximal tendon broken

fedeceola, 2023-11-27:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

As per title, while running some experiments the proximal tendon of the right thumb broke.

### Detailed context

-

### How does it affect you?

I cannot run my experiments and I have an upcoming deadline for a paper submission.

xEnVrE, 2023-11-27:
cc @sgiraz @GiacomoVassallo13 @AntonioConsilvio

GiacomoVassallo13, 2023-11-29:
@fedeceola @xEnVrE . The thumb proximal joint has been replaced and is ready to be tested. I attend a feedback.  

cc: @sgiraz @Fabrizio69

xEnVrE, 2023-11-30:
This has been tested. Closing.

cc @fedeceola

closed 2023-11-30 by xEnVrE

sgiraz, 2023-11-30:
Just to clarify, 

@AntonioConsilvio and I found the mais board of the right hand, non properly connected.

[image]

[image]

We have to pay attention to the wires when re-closing the covers (especially on the hands)

Cc @GiacomoVassallo13