robotology/mesh-support #1691: The left arm oscillates in certain configurations, probably due to the shoulder roll

xEnVrE, 2023-11-28:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

While testing a demo today I noticed that in certain configuration the left arm oscillates a lot making the robot possibly **unsafe**. At a certain point the should roll went in HF due to `i2t`.

FYI cc @enyen @GiuliaP 

### Detailed context

I was running a demonstration where the arm is controlled in Cartesian space. Along the planned trajectory there are configurations in which the arm oscillates a lot. The demonstration is not new and has been executed lots of times, hence we not it is safe. 

To me the problems seems very similar to what we had in https://github.com/robotology/icub-tech-support/issues/1488#issuecomment-1406316146. Hence, probably the encoder is not well connected.

### How does it affect you?

@gabrielecaddeo would need to use the left arm for a demonstration this Friday. Without the arm we can partially execute the demonstration. Thank you

GiacomoVassallo13, 2023-11-29:
Hi @xEnVrE . The left shoulder roll  locked motor has been restored. The malfunction  was caused to the loose srews of wave generator of harmonic drive. 
The screws has been thight and the harmonic was cleaned and regreased.
The joint it' s ready to test.

cc: @Fabrizio69 @sgiraz

xEnVrE, 2023-11-29:
Thanks @GiacomoVassallo13 

At the moment we cannot run the robot due to the following errors:

[image]

Attaching the full `yarprobotinerface` log just in case: 

[log_icub-head_yarprobotinterface_2141.txt](https://github.com/robotology/icub-tech-support/files/13501261/log_icub-head_yarprobotinterface_2141.txt)

cc @sgiraz @AntonioConsilvio

xEnVrE, 2023-11-29:
I tried again. It seems that the system detects the fault button as pressed - as if the fault line was interrupted somewhere:

[image]

Full log: 
[log_icub-head_yarprobotinterface_2521.txt](https://github.com/robotology/icub-tech-support/files/13501292/log_icub-head_yarprobotinterface_2521.txt)

sgiraz, 2023-11-29:
Hi @xEnVrE,

@AntonioAzocar is going to check for faulty cables... stay tuned!

xEnVrE, 2023-11-30:
We have tested the motion of the `shoulder roll` and it was fine. Closing.

Aside: the robot did not started because of a malfunctioning torso IMU. I am going to open a new issue specifically for that. For the moment being we are commenting the relevant part inside the `icub_all_no_legs.xml` file inside `$ROBOT_CODE/src/robotology_superbuild/src/robots-configuration/iCubGenova11`.

closed 2023-11-30 by xEnVrE