robotology/mesh-support #1692: Noise when moving the left shoulder roll with motors off

gabrielenava, 2023-11-29:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

While manually moving the left shoulder roll, I heard a noise and the motion seems discontinuous. The same does not happens on the right shoulder.

### Detailed context

For the moment, the joint starts correctly and moves. During the wind tunnel campaign we will monitor the situation. Maybe it is just necessary to clean the mechanism.

GiacomoVassallo13, 2024-02-07:
Hi. As agreed with @gabrielenava and @sgiraz  I will proceed to the shoulder surgery. 

cc: @Fabrizio69 @sgiraz

sgiraz, 2024-02-15:
The intervention has been suspended since @GiacomoVassallo13 will not be availble until the end of the month.

In the meantime @fbiggi may handle this issue. I'll give you updates asap.

cc @Fabrizio69 @AntonioConsilvio

gabrielenava, 2024-02-28:
just tested and the robot is working fine, to me we can close the issue. There was an issue at the startup phase, but I discovered it was the fault button that broke. I'll open a dedicated issue for that!

closed 2024-02-28 by sgiraz