robotology/mesh-support #1694: IMUs prevents the robot from running

xEnVrE, 2023-11-30:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

After #1691, the robot did not start properly.

@AntonioConsilvio noticed that by commenting the wrappers of the inertial sensors, i.e. the block https://github.com/robotology/robots-configuration/blob/5b7b79960d9edc5b4fa6b2418e6e8e96aab7cfea/iCubGenova11/icub_all_no_legs.xml#L51-L60

the robot starts properly.

Therefore, we think that one of the robot IMUs is creating issues. It requires further investigation.

### Detailed context

After #1691, the robot did not start properly.

@AntonioConsilvio noticed that by commenting the wrappers of the inertial sensors, i.e. the block https://github.com/robotology/robots-configuration/blob/5b7b79960d9edc5b4fa6b2418e6e8e96aab7cfea/iCubGenova11/icub_all_no_legs.xml#L51-L60

the robot starts properly.

Therefore, we think that one of the robot IMUs is creating issues. It requires further investigation.

xEnVrE, 2023-12-01:
Today I collected the log a `yarprobotinterface` run to ease debugging this:

[image]

As can be seen `embObjIMU` could not open for `waist-eb5-intertials`.

The full log follows: 

[log_icub-head_yarprobotinterface_2028.txt](https://github.com/robotology/icub-tech-support/files/13523509/log_icub-head_yarprobotinterface_2028.txt)

sgiraz, 2024-01-19:
Hello @martinaxgloria, I noticed that you recently used the robot. Have you encountered this issue again?

martinaxgloria, 2024-01-19:
Hi @sgiraz! I used the robot in the past days and I used the IMU mounted on the head for some tests, so I think that the block of code mentioned was not commented out. I didn't have this error and the yarprobotinterface worked properly. If I encounter it, I will re-open the issue

xEnVrE, 2024-01-19:
Hi all, please notice that the error above was about the `waist` imu not the head one.

martinaxgloria, 2024-01-19:
Hi @xEnVrE, thanks for the clarification. At this point, on Monday I'll have a look at the configuration file used to launch the yarprobotinterface and I'll see if the error is already present or not

cc @sgiraz

martinaxgloria, 2024-01-22:
Hi @sgiraz!

Today I looked at the conf files that the yarprobotinterface uses and the lines that enable the imu on the waist were enabled. I launched the robot with it and it started correctly.

sgiraz, 2024-01-23:
As the problem 

> Today I looked at the conf files that the yarprobotinterface uses and the lines that enable the imu on the waist were enabled. I launched the robot with it and it started correctly.

Nice one!
Thanks @martinaxgloria for your tests.
I proceed closing this issue.

>[!Note]
>Should the issue recur, we will reopen the issue and investigate the possible causes further

cc @AntonioConsilvio @Nicogene

closed 2024-01-23 by sgiraz