robotology/mesh-support #1697: Right elbow do not conclude the calibration

GiulioRomualdi, 2023-11-30:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The right elbow starts the calibration, reaches the limit, and goes into HF

### Detailed context

Here is the video of the calibration 

https://github.com/robotology/icub-tech-support/assets/16744101/b2e0e37e-d6e3-423a-b9ef-602665b4f8e6

AntonioConsilvio, 2023-12-01:
Hi @GiulioRomualdi, I tried disabling the AMO because it seemed to have some problem in reading the magnet, but mostly the joint is pretty hard even moving it by hand, especially when it reaches the hard limit during calibration.

@fbiggi and @elio75 are working on it, but in the meantime I have increased the PWM at startup so that the motor can calibrate.

sgiraz, 2024-01-04:
> Hi @GiulioRomualdi, I tried disabling the AMO because it seemed to have some problem in reading the magnet, but mostly the joint is pretty hard even moving it by hand, especially when it reaches the [...]

Well done @AntonioConsilvio 🚀,
The right elbow seems to calibrate correctly after these changes.

I consider this issue solved.

cc @GiulioRomualdi

closed 2024-01-04 by sgiraz