robotology/mesh-support #1698: Left wrist tendon broken

fedeceola, 2023-12-04:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

As per title, while running some experiments the the tendon of the left wrist broke.

### Detailed context

-

### Additional context

[image]

xEnVrE, 2023-12-11:
Dear all, there is a demonstration on December 20th for the OpenDay. Is there a plan for repairing the tendon of the wrist? Thank you

cc @sgiraz

GiacomoVassallo13, 2023-12-14:
Hi @xEnVrE . Today I replace the broken wrist tendon. Actually the joint work well, now need to test it under control.  All of proximal has been tense, because they are involved in the extraction of the wrist motor.
I attend feedback after testingon control.

cc: @Fabrizio69 @sgiraz

xEnVrE, 2023-12-14:
Hi @GiacomoVassallo13 I tested the robot now. Unfortunately it does not complete the calibration.

I am attaching the log of the `yarprobotinterface`: 
[log_icub-head_yarprobotinterface_1786.txt](https://github.com/robotology/icub-tech-support/files/13675501/log_icub-head_yarprobotinterface_1786.txt)

cc @sgiraz @AntonioConsilvio

sgiraz, 2023-12-14:
Ho @xEnVrE,

What about the motorgui readings?
Do you spot any encoder stuck to a fixed value or with a wrong reading?

@GiacomoVassallo13 @AntonioAzocar I would chack the encoders connected to the eb24:

[image]

xEnVrE, 2023-12-14:
I tried to restart it again and now it worked - not sure what happened, maybe one of the board did not start properly? I tested re-running it several time from scratch and it worked.

I think we can close the issue.

Thanks @GiacomoVassallo13 @sgiraz

closed 2023-12-14 by sgiraz