robotology/mesh-support #1700: Right thumb proximal tendon broken

fedeceola, 2023-12-04:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

As per title, while running some experiments the proximal tendon of the right thumb broke.

### Detailed context

As per title, while running some experiments the proximal tendon of the right thumb broke. I cannot run my experiments and I have an upcoming deadline for a paper submission.

### How does it affect you?

I cannot run my experiments and I have an upcoming deadline for a paper submission.

fedeceola, 2023-12-05:
Thanks @GiacomoVassallo13 @AntonioConsilvio @sgiraz !

fedeceola, 2023-12-05:
The tendon broke again. I attach a video (with the audio) of how it happened in case it may be useful.

https://github.com/robotology/icub-tech-support/assets/32268209/7a2b62a2-e88e-45cc-9c9a-126cd2241766

Fabrizio69, 2023-12-06:
cc: @GiacomoVassallo13

fedeceola, 2023-12-06:
Thanks @GiacomoVassallo13, I try using the robot as soon as possible.

GiacomoVassallo13, 2023-12-11:
@fedeceola have you got some feedback for me ?

fedeceola, 2023-12-11:
@GiacomoVassallo13, I used the robot last week. The thumb was working properly. Thanks!

closed 2023-12-11 by sgiraz