robotology/mesh-support #1702: Right thumb proximal tendon broken

fedeceola, 2023-12-12:
### Robot Name 🤖

None

### Request/Failure description

As per title, while running some experiments the proximal tendon of the right thumb broke.

### Detailed context

As per title, while running some experiments the proximal tendon of the right thumb broke. I cannot run my experiments and I have an upcoming deadline for a paper submission.

### How does it affect you?

I cannot run my experiments and I have an upcoming deadline for a paper submission.

fedeceola, 2024-01-22:
Hi, are there any plans to fix this? I'd need to do some tests by the end of the week.

sgiraz, 2024-01-23:
Hi @fedeceola,

Today @GiacomoVassallo13 started the maintenance activity on `iCubGenova11`, further updates will follow tomorrow morning

cc @AntonioConsilvio

fedeceola, 2024-01-24:
Hi, with @AntonioAzocar we just tried to run the robot. It cannot complete the calibration of the right thumb. The motor of the proximal joint is moving but the finger doesn't close. Furthermore, the tactile sensor of the thumb is not working.

cc @sgiraz @GiacomoVassallo13 @AntonioConsilvio

fedeceola, 2024-02-12:
Hi, actually I couldn't test the joint because, due to the issue in #1723, the right hand couldn't finish the calibration.

AntonioConsilvio, 2024-03-06:
Hi @fedeceola! Today I checked the status of the robot.

All tendons were intact, but the right thumb proximal joint didn't work because the connector was quite detached from the MC4-PLUS. 

Now all joints are working properly:

https://github.com/robotology/icub-tech-support/assets/114915464/2b54c7b3-6c5f-4d0c-9bfc-72cb78d86fa1

In addiction I fixed the configuration files, see the PR:

- https://github.com/robotology/robots-configuration/pull/626

cc @sgiraz

fedeceola, 2024-03-06:
Hi @AntonioConsilvio, thanks! I'll do some tests these days!

cc @lornat75

fedeceola, 2024-03-14:
Hi, yesterday and today I did some tests. The thumb is working. I think that we can close the issue. Thanks!

closed 2024-03-14 by sgiraz