robotology/mesh-support #1704: Robot shoulder wobbling after fall

elandini84, 2023-12-13:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

The robot left should shoulder is wobbling significantly if pushed or pulled by hand (even gently).

### Detailed context

The robot crate fell from the truck when the couriers where unloading it at Palazzo della borsa for the Edge 2023 event. After removing the robot from the crate, we checked it and found out about the shoulder. We then removed the torso covers to check if there were any visible damages. But we found none.
The robot should be checked in order to exclude any internal damage that could become worse moving the arm during demos or experiments.

### Additional context

This video shows the behaviour of the robot shoulder

https://github.com/robotology/icub-tech-support/assets/62991576/7f2c65c5-2a87-4af5-b6c9-fc5f7631cbf0

fbiggi, 2023-12-14:
Below are the various interventions carried out
after uncovering the robot I went to enter 2 code spacers
wingst 10850 AS-1528
V4-40--_-_U5931_GP screws have been registered
arm play has been restored
Movement test performed together with @ste93 : test ok

[image]

[image]
[image]
[image]

@maggia80 @sgiraz @Fabrizio69

elandini84, 2023-12-14:
Thanks a lot @fbiggi.
We can consider the issue closed.

closed 2023-12-15 by maggia80