robotology/mesh-support #1706: Left and right ankle roll motors moving

G-Cervettini, 2023-12-13:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The left and right ankle roll motors seem to move when pushed.  

### Detailed context

This is the linked video. If you enable the audio, you can listen to the sound produced by the ankles.

### Additional context

It occurred in the past as in https://github.com/robotology/icub-tech-support/issues/1662

### How does it affect you?

cc @DanielePucci  @GiulioRomualdi

DanielePucci, 2023-12-13:
We should understand a way to avoid this problem to recur. It may happen at demo time, and we may not spot the problem easily 

CC
@sgiraz @Fabrizio69 @maggia80

maggia80, 2023-12-15:
@fbiggi fixed the issue yesterday. The length of the screw was a bit too much touching the bottom of the threaded hole instead of tightening the two parts.

sgiraz, 2024-01-04:
> @fbiggi fixed the issue yesterday. The length of the screw was a bit too much touching the bottom of the threaded hole instead of tightening the two parts.

Thanks @fbiggi!
@fiorisi I suppose that some changes are needed on CAD, right? 

I consider this issue solved.

closed 2024-01-04 by sgiraz

fbiggi, 2024-01-05:
Hi, @sgiraz all changes to CAD have already been handled by Elio