robotology/mesh-support #1707: Broken cable on board 10.0.1.12

davidegorbani, 2023-12-15:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We were starting the robot but it was not working and the logger showed the following error:

[image]

So we opened the right lower leg and foot cover and found out that the cable connecting the front FT to the board is broken:

[image]

CC @gabrielenava 

### Detailed context

We broken the CAN2:1 cable from the `10.0.1.12` board to the FT sensor.

davidegorbani, 2023-12-15:
@AntonioAzocar fixed it and not the robot is working.

sgiraz, 2024-01-04:
Well done @AntonioAzocar!

Closing.

closed 2024-01-04 by sgiraz