robotology/mesh-support #1708: [Provide a meaningful TITLE, but DON'T enter the robot name here]

isorrentino, 2023-12-15:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The left hip roll joint goes into hardware fault as soon as the robot starts walking.

Here the video.

https://github.com/robotology/icub-tech-support/assets/43743081/5afd825b-8364-4bf4-a84b-090b789dac98

### Detailed context

Here is the log file

[log_ergocub-torso_yarprobotinterface_1154.txt](https://github.com/robotology/icub-tech-support/files/13685563/log_ergocub-torso_yarprobotinterface_1154.txt)

[image]

DanielePucci, 2023-12-16:
Most likely, the motor was burned after one-foot balancing experiments since that motor was pretty hot. This is another element for the importance of the temperature protection feature that is currently being implemented by iCub Tech, and that hopefully will be deployed onto the robot soon.

CC
@ale-git @valegagge

GiulioRomualdi, 2023-12-18:
Hi all, by any chance, if the motor is burned, is there another one to replace it?

maggia80, 2023-12-18:
@GiulioRomualdi yes we have two more motors left. The problem is that we have zero mech guys till Gen 2nd...

mebbaid, 2023-12-18:
We were trying an integration experiments with HSP maniupulation team using the walking controllers, but this issue is blocking.

fbiggi, 2024-01-03:
LLM1 engine replaced
work carried out see    [issue](https://github.com/icub-tech-iit/proto/issues/2680) PROTO
I deliver the robot to for testing to @AntonioConsilvio 

@Fabrizio69 @maggia80

maggia80, 2024-01-03:
@AntonioConsilvio we should change the current limit for this motor type **MOOG_C2900576_ROTOR** (see cub-tech-iit.github.io/documentation/ergoCub1/ergoCub1-joints) to:

```
 <param name="motorNominalCurrents">             5000                     </param>
        <param name="motorPeakCurrents">        10000               </param>
        <param name="motorOverloadCurrents">    15000           </param>
        <param name="motorPwmLimit">            18000               </param>
```

sgiraz, 2024-01-04:
A PR on `robot-configuration` with the changes mentioned above will follow soon.

cc @AntonioConsilvio

SimoneMic, 2024-01-22:
Hello,

Has this issue been solved?
Can we use the robot in its current state or do we need to take some further precautions?

Thanks!

AntonioConsilvio, 2024-01-22:
Hi @SimoneMic, As long as the temperature reading on the robot is not fully functional, the current limits will not change as it would cause the hip roll i2t HW fault during walking.

So, the usable current limits are still the same (devel).

sgiraz, 2024-03-14:
Hi @GiulioRomualdi,

To definitively close this issue we'd need someone who does the robot walking. 
Can you assist @AntonioConsilvio w/ this task?

cc @DanielePucci

AntonioConsilvio, 2024-05-29:
Hi! We tested [these current limits](https://github.com/robotology/icub-tech-support/issues/1708#issuecomment-1875082561) on ergoCubSN002:
> ````
> <param name="motorNominalCurrents">             5000                     </param>
>        <param name="motorPeakCurrents">        10000               </param>
>        <param [...]

The robot works but the joint tracking is less accurate, negatively affecting the walking. 

So more studies can be done on this to verify its feasibility, however I am proceeding with the closure of this issue as the burnt-out motor has been successfully replaced!

closed 2024-05-29 by AntonioConsilvio