robotology/mesh-support #1710: Fingers on the left-arm not working properly

ddetommaso, 2024-01-16:
### Robot Name 🤖

iCubErzelli01 S/N:032

### Request/Failure description

We have encountered issues on fingers of the left-arm, specifically related to thumb and pinky joints.

### Detailed context

During calibration the fingers often get stuck. 
Moreover, even when the calibration is successful, the little fingers are not moving from yarpmotorgui the interface (motor and encoders seem fine). 

### Additional context

[log_icub-head_yarprobotinterface_3025.zip](https://github.com/robotology/icub-tech-support/files/13949478/log_icub-head_yarprobotinterface_3025.zip)

[image]

https://github.com/robotology/icub-tech-support/assets/3440640/4bcdfdbd-5cdb-4a13-8a1d-7e7b2fd72d42

### How does it affect you?

High impact. The robot is currently in use for an experimental protocol and we needed to cancel participants.

AntonioConsilvio, 2024-01-19:
Hi Davide, a quick test. The solution may be simpler than expected. 
Specifically it is to make sure the configuration files are aligned on this robots-configuration master commit:

- https://github.com/robotology/robots-configuration/pull/596

In particular, this PR had originated from the last intervention performed to iCuberzelli01 (see [issue](https://github.com/robotology/icub-tech-support/issues/1648)).

Let us know as soon as you can!

cc @sgiraz

sgiraz, 2024-01-23:
Hi @ddetommaso,

The maintenance activity is scheduled for tomorrow!

cc @GiacomoVassallo13 @AntonioConsilvio

AntonioConsilvio, 2024-01-25:
Hi @ddetommaso. Yesterday me and @GiacomoVassallo13, after several tests, found the flat cable that connect the pinky motor the MC4-PLUS EB26 damaged. 
So, we cut off most of the damaged flat and soldered a new one.

The robot seemed to works properly! But we are waiting for feedback before proceeding with closing the issue.

We also found once that the thumb distal was not calibrating, but the problem was solved by restarting the motors. For now talking with @ddetommaso it was considered a non-urgent or blocking issue, so it was agreed not to investigate further for now.

cc @sgiraz

sgiraz, 2024-01-29:
Hi @ddetommaso,

Do you have any feedback to share w/ us?

ddetommaso, 2024-01-29:
Hi @sgiraz 

I am still waiting to provide our feedback for a couple of days of testing.

Thank you

ddetommaso, 2024-02-02:
Hi guys,

we can close this. The issue with the pinky is fixed

sgiraz, 2024-02-02:
Thanks for your feedback @ddetommaso.

Closing!

closed 2024-02-02 by sgiraz