robotology/mesh-support #1711: Update BAT board firmware

gabrielenava, 2024-01-16:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We are performing walking experiments on iRonCub-Mk3, but there are some difficulties. More specifically, the whole robot shuts down when one of the hip roll joints overcome the current limits.

@S-Dafarra told us that this routine has been improved in the more recent firmware versions of the BAT board. Would it be possible to align the BAT board firmware version with the one of ErgoCub?

### Detailed context

cc @DanielePucci @S-Dafarra @davidegorbani

S-Dafarra, 2024-01-17:
It might be worth to check also if the resistors on the board are the correct ones

AntonioConsilvio, 2024-01-18:
I updated the firmware with the last devel. 

Let me know if now everything work as expected.

gabrielenava, 2024-01-18:
everything is working fine @AntonioConsilvio!

at the beginning, the update of the BAT board alone was not enough to prevent the robot to sometimes shut down during walking. After also tuning the step width and length, we were able to walk repeatedly without any issue!

to me we can close, thanks!