robotology/mesh-support #1712: Fingers' positions are too much noisy

PasMarra, 2024-01-16:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Even when the fingers are kept still, their measured positions is varying too much over time because of the noisy measurements.

### Detailed context

Here you can find some useful logging file: https://istitutoitalianotecnologia-my.sharepoint.com/:f:/g/personal/pasquale_marra_iit_it/EiYZm4pUaRBMlSTIA5OfwaYBPdCQ2MF5ZT_LZ-7tUBccHQ?e=Duk9kV

### Additional context

The fingers on the real robot are not moving, yet this is what happens:

[fingers_trembling.webm](https://github.com/robotology/icub-tech-support/assets/143645988/a389faf9-71fd-4d3b-873e-b71b4d19a6e2)

Nicogene, 2024-01-18:
@PasMarra a curiosity, that visualization is rviz?
If yes, how are you publishing the decoupled fingers joints

PasMarra, 2024-01-18:
Hi @Nicogene, I've created a module to read the joint states from the `/joint_states` topic and, given the [parameters](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/gazebo_ergocub_left_hand_mk5_1.ini) and the [coupling laws](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling), I evaluate and publish a new message on a new topic (`/joint_states_full`), from which the `robot_state_publisher` can retrieve the full state.

Nicogene, 2024-03-19:
Hi @PasMarra!

we are aware that the measures of the fingers encoders may be affected by noise, it would be very useful if you can post a plot of the fingers encoder readings in order to understand the magnitude of the problem.

>,I've created a module to read the joint states from the /joint_states topic and topic and, given the [parameters](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/gazebo_ergocub_left_hand_mk5_1.ini?rgh-link-date=2024-01-18T14%3A51%3A59Z) and the [coupling laws](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling), I evaluate and publish a new message on a new topic (/joint_states_full

Moreover, if you point us to the software you are using (also privately in case it is not sharable), this would help debug the problem.

Thanks!

cc @maggia80

PasMarra, 2024-03-19:
Hi @Nicogene,
unfortunately I will not be back at CRIS earlier than April 8th and I've noticed that the attached rosbag

> Here you can find some useful logging file: https://istitutoitalianotecnologia-my.sharepoint.com/:f:/g/personal/pasquale_marra_iit_it/EiYZm4pUaRBMlSTIA5OfwaYBPdCQ2MF5ZT_LZ-7tUBccHQ?e=Duk9kV

 does not contain any encoders' measurements, only the `/joint_states` and '/joint_states_full' topics, so I cannot provide any of the plots you are asking for. 

I'm gonna send you privately the software I'm using to publish the full joints state.

Nicogene, 2024-03-19:
/remind 8th April Revive this activity

PasMarra, 2024-04-16:
Hi @Nicogene, these are the plots for each encoder when the fingers are kept still.

Notice that the `l_thumb_add` and `r_thumb_oc` signals are completely flat, maybe thay are not working at all.

# LEFT HAND
<p>
    [image]
    <em>Left hand</em>
</p>

<details>
<summary> LEFT FINGERS </summary>

<p>
    [image]
    <em>l_thumb_oc</em>
</p>

<p>
    [image]
    <em>l_thumb_add</em>
</p>

<p>
    [image]
    <em>l_index_oc</em>
</p>

<p>
    [image]
    <em>l_index_add</em>
</p>

<p>
    [image]
    <em>l_middle_oc</em>
</p>

<p>
    [image]
    <em>l_ring_pinky_oc</em>
</p>

</details>

# RIGHT HAND

<p>
    [image]
    <em>Right hand</em>
</p>

<details>
<summary> RIGHT FINGERS </summary>

<p>
    [image]
    <em>r_thumb_oc</em>
</p>

<p>
    [image]
    <em>r_thumb_add</em>
</p>

<p>
    [image]
    <em>r_index_oc</em>
</p>

<p>
    [image]
    <em>r_index_add</em>
</p>

<p>
    [image]
    <em>r_middle_oc</em>
</p>

<p>
    [image]
    <em>r_ring_pinky_oc</em>
</p>

</details>

Nicogene, 2024-04-16:
Thanks @PasMarra ! Are those plots in rad ?

PasMarra, 2024-04-16:
> Thanks @PasMarra ! Are those plots in rad ?

Yes, they are.

maggia80, 2024-04-16:
The TLV493D has this accuracy in the non-calibrated mode (we are using non-calibrated mode):

[image]

taken from: https://www.infineon.com/dgdl/Infineon-3D_Magnetic_Sensor_for_Angle_Measurements-AN-v01_10-ApplicationNotes-v01_01-EN.pdf?fileId=5546d46265f064ff01665a3d22f055e3

cc: @marcoaccame

maggia80, 2024-04-16:
the magnet used is this one: https://www.magfine.it/it/products/detail.php?product_id=22298

lornat75, 2024-04-16:
would you expect the error reported in your plot to correspond to variable noise as pasquale reported? it is weird that some encoders/fingers have different level of noise.

maggia80, 2024-04-17:
Channels with more noise could be related to the distance of the magnet from the sensor (we should check this correlation). 
It is worth investigating and seeing if we can reduce this noise.

simeonedussoni, 2024-05-13:
hi @PasMarra 
given that the forearms have been reworked, this effect is still present? can we have a hands-on with the robot to investigate the origin of noise?

PasMarra, 2024-05-13:
> hi @PasMarra given that the forearms have been reworked, this effect is still present? 

Hi @simeonedussoni, I haven't had the opportunity to check it.

> can we have a hands-on with the robot to investigate the origin of noise?

Yes, sure. Let's arrange a lab session on teams.