robotology/mesh-support #1714: Left camera not working in some configurations

paliasgh, 2024-01-18:
### Robot Name 🤖

iCubWaterloo01 S/N:044

### Request/Failure description

The left eye camera does not work in some configurations of the eye bulb.

### Detailed context

Hello, we are facing an issue similar to #1150 with the left camera of our robot. It has become worse recently so that sometimes we do not have any image after the robot startup (due to eyes rotating during startup calibration). We opened the face cover, and yes, the flat cable seems to be damaged. It is badly twisted near the connector on the camera (see picture below). Just holding the cable in a certain position fixes the problem and we can have the image again. Do you recommend we replace the cable? If yes, what are the specifications of this cable so that we can buy a replacement? Can the issue be related to the connectors? Thanks a lot.

[image]

maggia80, 2024-01-19:
Hi @paliasgh , yes the flat cable is the: Molex 98266-0215

you should be able to buy it here: https://www.mouser.ca/ProductDetail/Molex/98266-0215?qs=5aQ9wn8iLe%2FeRfDMj9hHuw%3D%3D&_gl=1*mse117*_ga*MTQ4ODI3NzU2My4xNzA1NjUwODk3*_ga_15W4STQT4T*MTcwNTY1MDg5Ny4xLjAuMTcwNTY1MDg5OS41OC4wLjA.*_ga_1KQLCYKRX3*MTcwNTY1MDg5Ny4xLjAuMTcwNTY1MDg5OS4wLjAuMA..

paliasgh, 2024-01-23:
Hello @maggia80 ,

The issue is solved with replacing the cable. Thanks.

closed 2024-01-23 by paliasgh