robotology/mesh-support #1717: Tons of errors when launching the robotinterface

DanielePucci, 2024-01-19:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The terminal is overwhelmed with errors

### Detailed context

[image]

### Additional context

@maggia80 @simeonedussoni @accame

simeonedussoni, 2024-01-19:
ciao @DanielePucci this issue is known and we are already working on it.

Our first step will be to align the robot to the last firmware set and check again. 

The very first messages of this log are pretty important to trace back the problem.
cc @marcoaccame

DanielePucci, 2024-01-19:
Ok great, @simeonedussoni when is the new firmware supposed to be on the robot? 

It's really difficult to use the robot now since the terminal cannot be checked for errors - we are having this problem now

simeonedussoni, 2024-01-19:
ciao @DanielePucci 

I can go and load the official firmwares as soon as I finish a WG brainstorming with the WRISTASK-FORCE ™️

DanielePucci, 2024-01-19:
After the update of firmware, the `EOtheInfoDispatcher` error disappeared. Thanks @simeonedussoni, closing.

closed 2024-01-19 by DanielePucci

marcoaccame, 2024-01-19:
> ciao @DanielePucci
> 
> I can go and load the official firmwares as soon as I finish a WG brainstorming with the WRISTASK-FORCE ™️

@DanielePucci : just to explain... on Wednesday afternoon just  before the start of the demo we saw that the right wrist of `ergoCub `was emitting this annoying  burst of messages but we decided to keep the situation like that.

the loaded binaries it was the best solution that ensured no hw fault occurrence but also the possibility to recover from a possible hw fault w/out powering down. 

after that the binary of the amc has stayed like that. 

however, we already have in devel the final binaries and @simeonedussoni will soon update the robot.
- https://github.com/robotology/icub-firmware-build/pull/131
- https://github.com/robotology/icub-firmware-build/pull/130 and https://github.com/robotology/icub-firmware-build/pull/132

About the `EOtheInfoDispatcher` messages: they are emitted when there are too many diagnostic errors and some of them are lost.  And yes, they are quite annoying as they are emitted every millisecond.

They may be due to a quick update of the boards that has wrongly left some extra diagnostic debug messages. 

So, @simeonedussoni pls let me know if the messages from the `EOtheInfoDispatcher` have disappeared after the alignment to devel.

simeonedussoni, 2024-01-19:
hi @marcoaccame 

I rebased the boards with the `devel` version of FW and the tons of messages disappeared.  There remains other (correct) tons of messages from the diagnostic flow.