robotology/mesh-support #1718: Problems when loading configuration for torque control parameters

DanielePucci, 2024-01-19:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Today, we started with @isorrentino to work to test the Yoga demo on the real robot after the preliminary results in https://github.com/ami-iit/element_qp-reactive-control/issues/70. 

So, we started modifying the files for loading the parameters of torque control. See the commit https://github.com/ami-iit/robots-configuration/commit/f6fe83cf51cf5e385477fbe4fdc2bd043313191e

Unfortunately, these parameters seem not loaded in the motor gui 

[image]

And we cannot at the moment test the torque control since adding them manually for all the joints is very 

### Detailed context

@Nicogene @sgiraz @maggia80

sgiraz, 2024-01-19:
Hi @DanielePucci,

The `Parser` seems to parse correctly the values set in config files:

```console
...

(stictionUp 0 0 0) (stictionDown 0 0 0) (kff 1 1 1) (viscousPos 0 0 0) (viscousNeg 0 0 0) (coulombPos 0 0 0) (coulombNeg 0 0 0) (velocityThres 0.1 0.1 0.1) (filterType 0 0 0) (ktau 176 176 -176)"), ("OTHER_CONTROL_PARAMETERS" [group] = "(deadZone 0.0049 0.0049 0.0049)")]

...
```

I'm not sure, but as a first quick view to the info provided, it could be a (wrong) cast from float to int that trunkates the 0.1 to 0. 
By the way,  we'll perform some tests in during our next sprint (starting from Monday 22).

cc @Nicogene

sgiraz, 2024-01-22:
Hi @DanielePucci,

I fixed a bug in `icub-main` that was preventing the parsing of torque parameters when the (⚠️ deprecated) `kbemf` parameter was missing. Here's the PR that addresses the issue:
- https://github.com/robotology/icub-main/pull/935

cc @isorrentino @GiulioRomualdi @AntonioConsilvio

DanielePucci, 2024-01-22:
Thanks a lot @sgiraz, hopefully tomorrow we are going to give it a try!

isorrentino, 2024-01-24:
Today, @DanielePucci and I tested the PR. Now everything works fine. Thanks a lot.

sgiraz, 2024-01-25:
Hi @isorrentino, @DanielePucci 

Thank you for your precious feedback.

I proceed closing this issue! ✅

Cc @Nicogene @AntonioConsilvio

closed 2024-01-25 by sgiraz