robotology/mesh-support #1719: Error when launching the yarpmotorinterface on

DanielePucci, 2024-01-19:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

We are getting the following error when launching the robotinterface

~~~
 5.308596 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=258s 946m 170u : SYS: a service has detected that some CAN boards have stopped transmission. Type of service category is eomn_serv_category_all. Lost CAN boards are on (can1map, can2map) = ([ 2  ], [  ]). Time since last contact: 0 [ms]
~~~

 5.308596 <ERROR> from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=258s 946m 170u : SYS: a service has detected that some CAN boards have stopped transmission. Type of service category is eomn_serv_category_all. Lost CAN boards are on (can1map, can2map) = ([ 2  ], [  ]). Time since last contact: 0 [ms]

### Detailed context

[log_ergocub-torso_yarprobotinterface_18098.txt](https://github.com/robotology/icub-tech-support/files/13989678/log_ergocub-torso_yarprobotinterface_18098.txt)

sgiraz, 2024-01-29:
It may be caused by https://github.com/robotology/icub-tech-support/issues/1715#issuecomment-1903590104

cc @AntonioConsilvio @maggia80 @MSECode @Fabrizio69

MSECode, 2024-01-29:
I think it is not related to BAT since the CAN board lost are on CAN1 as shown by the error log and BAT is on CAN2.

MSECode, 2024-01-29:
**Errata**: I checked the configuration files and I saw that the ems board eb2 is connected to the BMS and not to the BAT, which is instead connected to the right_arm_eb1. Anyway, as for BAT, also the BMS is configured on CAN2:2, so you are having errors on the 2FOC, which are actually configured at CAN1:1:0 and CAN1:2:0

valegagge, 2024-01-29:
I saw the log today, I think that @MSECode was right, but, in any case,  we need to investigate why the diagnostic says: `Time since last contact: 0 [ms]`.

valegagge, 2024-01-29:
I opened this issue for investigate further. https://github.com/robotology/icub-main/issues/937

sgiraz, 2024-03-06:
Hi @valegagge,

As per https://github.com/robotology/icub-main/issues/937#issuecomment-1961691892, I guess We can close this issue, right?

cc @MSECode

valegagge, 2024-03-07:
Hi @sgiraz,
in https://github.com/robotology/icub-main/issues/937, we fixed the diagnostic messages. We didn't take care of the problem on the robot.
Now with the new release the diagnostic provides correct information when a CAN board stops transmitting.

In this ticket, it seems that a 2FOC stops transmitting.

Before closing this issue, I suggest verifying if users experienced this problem again. 

Thanks!

AntonioConsilvio, 2025-02-20:
Hi! This error is still present on the robot.
It does not seem to show any actual malfunctions, but the error continues to be displayed.
It could be due to the `bms` board that is connected to that CAN, but I don't actually know for sure.

MSECode, 2025-02-20:
> Hi! This error is still present on the robot. It does not seem to show any actual malfunctions, but the error continues to be displayed. It could be due to the `bms` board that is connected to that CAN, but I don't actually know for sure.

It is most likely due to that. Anyway the error string should be different as specified here:
- instead of `eomn_serv_category_all` you should have `eomn_serv_category_battery`
- instead of  `(can1map, can2map) = ([ 2 ], [ ])` you should have `(can1map, can2map) = ([ ], [ 13 ])` since bms is on the CAN2:2 and it is of class 13 (if I recall correctly)
- and for the time you should actually have a time different from 0

Just a note. The error message will be correct only if the bms is configured correctly in the configuration files and if the icub-main version comes from 2 release ago or more recent. 
I'm quite confident that if you connect the bms the error will disappear. 
If this is not the case and the error string is still the same as in the body of the issue we should investigate more.
Thanks

MSECode, 2025-02-21:
For now to avoid spamming the warning you can use the disable tag `disable_battery_bms` (the tag should be update if possible since now we have just a general tag for the battery service, which is `disable_battery`, so that you do not need to change the robots configuration files every time. 
In the meanwhile we are going to work on a smart fw solution.
Thus I think we can close this issue now. 
I'll keep this noted w/ @valegagge but for now feel free to close the issue if solved.

closed 2025-03-05 by AntonioConsilvio