robotology/mesh-support #1720: ergoCub1.1 S/N:001 – during a Demo the wrist stopped movingYesterday during a Demo the wrist stopped moving. Even if the motor was not controllable the motorgui indicated that the joints were in position direct. Moving back the joints in position did not solve the problem.

sgiraz, 2024-01-19:
Yesterday during a Demo the wrist stopped moving. Even if the motor was not controllable the motorgui indicated that the joints were in position direct. Moving back the joints in position did not solve the problem.

Please notice that we use to control the wrist in positiondirect.

_Originally posted by @GiulioRomualdi in https://github.com/robotology/icub-tech-support/issues/1709#issuecomment-1900113622_

sgiraz, 2024-01-19:
We recently deployed an updated firmware incorporating the latest changes from our development branch. While this firmware introduces notable enhancements, we cannot fully confirm its effectiveness in resolving this specific issue.

We are continuing our investigation into the root cause of this issue. In the meantime, please continue monitoring the situation for any further developments.

cc @valegagge @simeonedussoni @marcoaccame @AntonioConsilvio @GiulioRomualdi

GiulioRomualdi, 2024-01-19:
Thank you, @sgiraz! Yes, we will check the wrist every time we work on the robot!

simeonedussoni, 2024-01-22:
I ask the users to save logs with the `yarplogger` in case the problem reappears and to attach them here for an easier investigation

cc @GiulioRomualdi 

I'm putting this advice also on ergocub channel

valegagge, 2024-01-23:
Next step:
 - check the configuration on robot
 - tests: start and stop YRI on the wrist on bench logging YRI and CAN messages, checking if it stops or amc2c disappears.

simeonedussoni, 2024-01-25:
I'd have a F2F discussion and hands on w/ @GiulioRomualdi @S-Dafarra together with at least @valegagge to investigate some items on the code.

our strategy is
- [ ] take data with the teleoperation which seems to trigger the bug
- [ ] record the setpoints of at least the wrist joints
- [ ] understand at which rate they are sent and if this can cause unstable operation
- [ ] reproduce the bug on our setup possibly with the `WholeBodyPlayer` acting on the standalone wrist

_optionally_ we can add the `cansniffer` tool by attaching a derivation for the signals to study the ultra-low-level behavior of the board tree

In order to have info to continue our investigation it is deally to be held today, Jan. 25th

simeonedussoni, 2024-01-25:
Some conversation with @S-Dafarra and @GiulioRomualdi gave the following outcome

- data rate: the setpoints are refreshed with $500 Hz$ rate, targeting $1 kHz$
- the data dump is enable only when in `teleoperated walking `
- their code does not refresh the joint `command mode`

moreover, their feeling is it can be a similar problem to one faced in the past dealing with `MC4+` boards.

I kindly ask @GiulioRomualdi and @S-Dafarra to link the related issue here if they can retrieve it

simeonedussoni, 2024-01-30:
as of most recent news, @GiulioRomualdi and coworkers used teleop yesterday, **without experiencing this problem**

simeonedussoni, 2024-02-01:
I succeded in reproducing the error in the wrist setup. It seems related to a `timeout` occurring during the `home` procedure, it's unclear if related to the startup only or can be triggered also during operation. in this case it's likely that the control loop takes prioirty in deciding the motion parameters and thus the problem is not present. Investigations are under way.

simeonedussoni, 2024-02-07:
besides, I am testing the new Wrist with hardstops preventing the mechanism to flip around its swivels and go in reverted configuration 

https://github.com/robotology/icub-tech-support/assets/31098248/affc74f4-d4f2-4886-8762-c5d50b1dcb3f

simeonedussoni, 2024-03-20:
the hardstops are implemented on the SN002.
they will be also upgraded on SN001.
all fw bugs have been fixed:
- [PR](https://github.com/robotology/icub-firmware/pull/467)
- [PR](https://github.com/robotology/icub-firmware/pull/470)
- [PR](https://github.com/robotology/icub-firmware-build/pull/138)

closing

closed 2024-03-20 by simeonedussoni