robotology/mesh-support #1721: Neck cable broken

DanielePucci, 2024-01-24:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

During the tests for the yoga, the robot was faulted and then the neck cable got broken

### Detailed context

See above

fbiggi, 2024-01-25:
@DanielePucci 

Neck cord repaired
I await your info after movement tests

@Fabrizio69 @sgiraz @maggia80

GiacomoVassallo13, 2024-01-25:
We waiting the feedback of testing in control.

cc: @Fabrizio69 @sgiraz

sgiraz, 2024-01-29:
Hi @DanielePucci 

The neck has been successfully repaired by @fbiggi. The robot's current utilization suggests that the repaired neck is functioning properly, is it right?

cc @GiacomoVassallo13 @Fabrizio69 @AntonioConsilvio

sgiraz, 2024-02-01:
Nice! 

Closing as solved, then! 

>[!Note]
>Investigations and improvements are under the way in order to avoid this issue asap.

closed 2024-02-01 by sgiraz